exelban/stats #448: NordVPN connection unknown network interface

altrdev, 2021-04-29:
**Describe the bug**
When I connect with VPN the Interface, Network and Physical address disappears.
Country code disappear too:

[image]

**Details:**
 - Device: Macbook Pro 16 2018 and MacBook Pro 13 M1
 - MacOS 11.3
 - Application version 2.5.11

exelban, 2021-04-29:
Hi. Thanks for reporting.
The country code will be only if it's available.
The system creates a new interface for VPN (on some level). And some parameters will be unavailable. For example, there will be no physical address, because of the software interface. And the network is available only throw a WiFi connection.
I few some usage of VPN with Stats, and it works fine. It looks like NordVPN works in a different way. I cannot do anything with this.

altrdev, 2021-04-29:
Thanks for explanation. I reported because with iStatMenu network interface still wifi or eth when vpn is running.

exelban, 2021-04-29:
You can manually change the interface in the settings. By default, it selects the primary interface.

altrdev, 2021-04-29:
Yes it's an option. It's annoying if you switch wifi/eth very often.

Anyway if I choose wifi for example, I have wrong country code

[image]

The right code is Argentina

exelban, 2021-04-29:
Country code is fetching from macOS.

altrdev, 2021-04-30:
What do you think about substitute `https://api.ipify.org` with `https://freegeoip.app/json/` ?
We can collect public ip and right country code too.

exelban, 2021-04-30:
Hi. The `freegeoip.app` is on some pi-hole list. And I cannot use this one. And all users which use pi-hole could have the same problem too.

altrdev, 2021-04-30:
Any other apis? Like `https://ipapi.co/json`?

exelban, 2021-05-02:
I don't really understand why I need to change the service for fetching public IP.
It fetches the IP - yes. It was noticed in some block list - no.

closed 2021-05-02 by exelban

altrdev, 2021-05-02:
Sorry man, is not my idea make you angry.

Is useful, over VPN connection, show right country code. 
But is your project, no problem