exelban/stats #449: Feature request: Set the display order

kevindiu, 2021-04-30:
I think it would be nice to let user to set the display order of items.

For example user can drag the order of the settings here to change the display order on the menu bar.
[image]

exelban, 2021-04-30:
Hi. Thanks for the proposal. But it's not possible. Menu bar could be organized only by user.

closed 2021-04-30 by exelban

kevindiu, 2021-04-30:
Thank you for your answer. Sorry that I didn't know that before. I thought that it is managed by the application.

After googling for it, we can change the order by clicking the Command key and drag the icon.