exelban/stats #450: Feature request: Show system on time on battery item.

kevindiu, 2021-04-30:
It would be nice if the system on time shows on the battery drop-down detail.
[image]

exelban, 2021-04-30:
Hi. Thanks for the proposal. I think you mean time from last boot. If yes - it could be found in the dashboard.

kevindiu, 2021-04-30:
Yes I found it on the dashboard, but I think it may be more convenient if we can see it there.

kevindiu, 2021-04-30:
Also it may be useful if we show the `System on time since last charge".
I can help translate it to Traditional Chinese and Japanese.

exelban, 2021-04-30:
The last boot cannot be placed in the battery module. Because there are macs without batteries. And these users want to see this parameter (I want to see).

System on time since the last charge - it's a little bit hard to determine. It cannot be read from the system. It can be easily implemented in the Stats. But it will be more "since last power adapter".

kevindiu, 2021-04-30:
thank you for your quick response. 
For Macs without battery, we can consider to hide it on the menu.

Yeah I think `since last power adapter` is good enough.

BTW just an idea, for `System on time since the last charge`, we can implement it by writing a time to the database every minute, and base on the time the data written, we can calculate the system on time.

closed 2021-05-09 by exelban