exelban/stats #451: RX 5500 not reporting temps, fan speed, utilization

goodspeed, 2021-05-03:
Can anyone else confirm this? 

The MSI RX 5500 XT is showing 0 values for temp, fan speed & GPU utilization. 

It worked for our other setup RX 580.

exelban, 2021-05-03:
Hi. Could you please post the output of `ioreg -r -d 1 -w 0 -c "IOAccelerator"`?
It will help to debug if there is not this data in driver or something goes wrong in the Stats.

goodspeed, 2021-05-12:
Sorry I lost access to the machine. Here's the output--

ioreg -r -d 1 -w 0 -c "IOAccelerator"

+-o AMDRadeonX6000_AMDNavi14GraphicsAccelerator  <class AMDRadeonX6000_AMDNavi14GraphicsAccelerator, id 0x100000461, registered, matched, active, busy 0 (317 ms), retain 118>
    {
      "AccelCaps" = 15
      "IODVDBundleName" = "AMDRadeonVADriver2"
      "IOGVAHEVCEncode" = "1"
      "ATY,intrev" = 21
      "AccelNativeDMARowByteAlignment" = 256
      "InternalStatisticsAccm" = {}
      "IOGeneralInterest" = "IOCommand is not serializable"
      "GPUDCCDisplayable" = Yes
      "InternalStatistics" = {}
      "MetalPluginClassName" = "AMDMTLGFX10Device"
      "GpuDebugPolicy" = 0
      "AGCInfo" = {"fBusyCount"=0,"poweredOffByAGC"=No,"fSubmissionsSinceLastCheck"=150996,"fLastSubmissionPID"=188}
      "dpm" = 1
      "MetalPluginName" = "AMDRadeonX6000MTLDriver"
      "PerformanceStatisticsAccum" = {"iosurfaceTextureCreationBytes"=0,"HWChannel SDMA0_PAGE | Commands Completed"=0,"HWChannel VCN0Dec | Commands Completed"=0,"Device Utilization %"=0,"orphanedNonReusableSysMemoryCount"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=0,"oolTextureCreationCount"=0,"surfaceCount"=18,"HWChannel KIQ | Commands Submitted"=0,"HWChannel SDMA1_PAGE | Commands Completed"=0,"context2DCount"=4,"surfaceCopyOutWaitTime"=0,"surfaceReadLockIdleWaitTime"=0,"stdTexturePageInBytes"=0,"swapBytesPerSample"=0,"finishCLWaitTime"=0,"gartCacheBytes"=33554432,"agpTextureCreationCount"=0,"stdTextureCreationBytes"=0,"textureCount"=2577,"HWChannel VCN0Enc | Commands Submitted"=0,"surfaceTextureCreationCount"=0,"agprefTextureCreationBytes"=0,"finish2DWaitTime"=0,"contextVideoCount"=0,"orphanedNonReusableSysMemoryBytes"=0,"surfaceBufferPageOutBytes"=0,"swapComplete2DWaitTime"=0,"contextGLCount"=2,"surfaceBufferTextureCreationCount"=0,"orphanedReusableSysMemoryBytes"=0,"orphanedNonReusableVidMemoryCount"=0,"HWChannel GFX | Commands Completed"=0,"clientSharedAllocatedBytes"=1290240,"HWChannel Compute1 | Commands Completed"=0,"clientGLWaitTime"=0,"freeSurfaceSwapBufferWaitTime"=0,"Fan Speed(%)"=0,"surfaceBufferReadOutBytes"=0,"volatileSurfaceCount"=0,"hardwareSubmitWaitTime"=0,"HWChannel SDMA1 | Commands Submitted"=0,"HWChannel Compute1 | Commands Submitted"=0,"inUseVidMemoryBytes"=435888128,"vramFreeBytes"=7461748736,"ioSurfacePageInBytes"=0,"HWChannel SDMA0 | Commands Submitted"=0,"finishAll2DWaitTime"=0,"recoveryCount"=0,"orphanedReusableVidMemoryCount"=576,"Core Clock(MHz)"=0,"orphanedNonReusableVidMemoryBytes"=0,"HWChannel SDMA0_PAGE | Commands Submitted"=0,"Memory Clock(MHz)"=0,"vramEvictionWaitTime"=0,"stdTextureCreationCount"=0,"gartSizeBytes"=6442450944,"GPU Activity(%)"=0,"HWChannel VCN0EncLLQ | Commands Submitted"=0,"oolTexturePageInBytes"=0,"texturePageOutBytes"=0,"freeToAllocGPUAddressWaitTime"=0,"gartMapInBytesPerSample"=0,"Fan Speed(RPM)"=0,"HWChannel VCN0Dec | Commands Submitted"=0,"HWChannel VCN0Enc | Commands Completed"=0,"oolTextureCreationBytes"=0,"finishGLWaitTime"=0,"HWChannel KIQ | Commands Completed"=0,"HWChannel SDMA1 | Commands Completed"=0,"freeDataBufferWaitTime"=0,"HWChannel SDMA1_PAGE | Commands Submitted"=0,"orphanedReusableVidMemoryHitRate"=96,"HWChannel SDMA0 | Commands Completed"=0,"inUseSysMemoryBytes"=132866048,"hardwareWaitTime"=0,"bufferSwapCount"=0,"swapCompleteGLWaitTime"=0,"agprefTextureCreationCount"=0,"agpTextureCreationBytes"=0,"dataBytesPerSample"=0,"HWChannel GFX | Commands Submitted"=0,"HWChannel Compute0 | Commands Completed"=0,"iosurfaceTextureCreationCount"=0,"bufferFlipCount"=0,"gartMapOutBytesPerSample"=0,"Temperature(C)"=0,"contextCLCount"=0,"surfaceBufferPageInBytes"=0,"freeSurfaceBackingWaitTime"=0,"HWChannel Compute0 | Commands Submitted"=0,"textureVolunteerUnloadBytes"=0,"ioSurfaceReadOutBytes"=0,"finishVideoWaitTime"=0,"HWChannel Compute2 | Commands Completed"=0,"gartUsedBytes"=132866048,"Total Power(W)"=0,"textureReadOutBytes"=0,"orphanedReusableSysMemoryCount"=0,"gartFreeBytes"=6309584896,"dataBufferCount"=0,"ioSurfacePageOutBytes"=0,"orphanedReusableVidMemoryBytes"=675520512,"surfaceSetShapeIdleWaitTime"=0,"HWChannel Compute2 | Commands Submitted"=0,"surfaceWriteLockIdleWaitTime"=0,"surfaceCopyInWaitTime"=0,"HWChannel VCN0EncLLQ | Commands Completed"=0}
      "PerformanceStatistics" = {"iosurfaceTextureCreationBytes"=90112,"HWChannel SDMA0_PAGE | Commands Completed"=1059,"HWChannel VCN0Dec | Commands Completed"=0,"Device Utilization %"=7,"orphanedNonReusableSysMemoryCount"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=0,"oolTextureCreationCount"=55,"surfaceCount"=18,"HWChannel KIQ | Commands Submitted"=0,"HWChannel SDMA1_PAGE | Commands Completed"=148,"context2DCount"=4,"surfaceCopyOutWaitTime"=0,"surfaceReadLockIdleWaitTime"=0,"stdTexturePageInBytes"=87244800,"swapBytesPerSample"=0,"finishCLWaitTime"=0,"gartCacheBytes"=33554432,"agpTextureCreationCount"=11,"stdTextureCreationBytes"=113197056,"textureCount"=2577,"HWChannel VCN0Enc | Commands Submitted"=0,"surfaceTextureCreationCount"=0,"agprefTextureCreationBytes"=0,"finish2DWaitTime"=0,"contextVideoCount"=0,"orphanedNonReusableSysMemoryBytes"=0,"surfaceBufferPageOutBytes"=0,"swapComplete2DWaitTime"=0,"contextGLCount"=2,"surfaceBufferTextureCreationCount"=0,"orphanedReusableSysMemoryBytes"=0,"orphanedNonReusableVidMemoryCount"=0,"HWChannel GFX | Commands Completed"=713,"clientSharedAllocatedBytes"=1290240,"HWChannel Compute1 | Commands Completed"=3,"clientGLWaitTime"=0,"freeSurfaceSwapBufferWa [... cut ...] use video memory"),(103,6442450945,"Command Queue Count"),(104,6442450945,"GPU Restart Count")),"IOReportGroupName"="Internal Statistics","IOReportChannelInfo"={"IOReportChannelUnit"=0}})
      "IOProbeScore" = 0
      "IOGVAHEVCEncodeCapabilities" = {"VTPerProfileDetails"={"1"={"VTMaxEncodeLevel"=153}},"VTQualityRating"=50,"VTRating"=350,"VTSupportedProfileArray"=(1)}
      "IOAccelDisplayPipeCapabilities" = {"DisplayPipeSupported"=Yes,"TransactionsSupported"=Yes}
      "SurfaceList" = ()
    }

goodspeed, 2021-05-13:
hi @exelban, can you reopen this issue? I don't want to post a new issue. 

I replied above.

Thanks.

exelban, 2021-05-13:
Hi. Thanks for sharing the output.
According to the output, temperature and utilization must be available.
Did the RX 5500 is enabled?

goodspeed, 2021-05-13:
Yes, the RX 5500 is installed. 

This is on the hackintosh. I have the AMD Power Gadget working with AMDRyzenCPUPowerManagement.kext & SMCAMDProcessor.kext. 

It was working normally with the other hackintosh Radeon RX 580. But when I setup on another machine with the RX 5500XT, the GPU temp, fan speed & util have no values. 

What can I check or enable?

exelban, 2021-05-13:
Nothing. I will make a build with a log to file. It will allow debugging the problem.

goodspeed, 2021-05-13:
Thanks so much for your great work.

exelban, 2021-05-18:
Hi. Could you please run this build for a few minutes?
After you will find a file `log.txt` in the `Documents` folder. Could you please share this file?

[Stats.dmg.zip](https://github.com/exelban/stats/files/6503023/Stats.dmg.zip)

goodspeed, 2021-05-19:
Here it is.

[log.txt](https://github.com/exelban/stats/files/6511763/log.txt)

goodspeed, 2021-05-19:
[image]

What is missing are temp, fan speed, core clock, mem clock (not utilization)

exelban, 2021-05-20:
Ok. Your driver just not provide this information about GPU. Not all drivers provide the right stats. In your case driver have these stats in the API, but all have 0 value. There is no way I can fix it.

closed 2021-05-20 by exelban

goodspeed, 2021-05-20:
Thanks @exelban