exelban/stats #452: Sensors Widget does not show up correctly

wfrank21, 2021-05-03:
Since the last update, the sensors widget got cut slightly off. 
[image]
Plus, when clicking on this widget, the popout does not open.
I tried reopening the app and reactivating the widget, but that does not help.

**Details:**
 - Device: MacBook Pro, 13-inch, Four Thunderbolt 3 Ports
 - MacOS 11.3
 - Application version 2.5.12

wfrank21, 2021-05-03:
I have just realized that the Fans Widget is also cut off, but the popout still opens. 
[image]

exelban, 2021-05-03:
Hi. Thanks for reporting. I will try to fit the width for these values.

exelban, 2021-05-03:
Hi. I have made a fix for the sensors. 
Fans widget will be now without `RPM` if label value is enabled.

closed 2021-05-05 by exelban