exelban/stats #453: Stats 2.5.12 crash on M1 Macbook Air

lanthean, 2021-05-03:
**Describe the bug**
New version of Stats (2.5.12) crashes on M1 version of MBA (Macbook Air).
After a few moments after starting the app, it will crash.
I have SSD (R/W, F/U, usage %), RAM, CPU, Temp, Battery and network (U/D) enabled.

**Details:**
 - Device: Macbook Air 2020
 - MacOS 11.3 (20E232)
 - Application version 2.5.12
 
```sh
 ✔ lanthean@mba ➤ ~/mnt ➤ /Applications/Stats.app/Contents/MacOS/Stats                                  
Error call(READ_KEYINFO): (iokit/common) invalid argument
Error read(): (iokit/common) invalid argument
Error call(READ_KEYINFO): (iokit/common) invalid argument
Error read(): (iokit/common) invalid argument
2021-05-03 14:16:10.393 Stats[27147:7083104] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]'
*** First throw call stack:
(
        0   CoreFoundation                      0x000000018a8d3db8 __exceptionPreprocess + 240
        1   libobjc.A.dylib                     0x000000018a5fd0a8 objc_exception_throw + 60
        2   CoreFoundation                      0x000000018a99ebb4 -[__NSCFString characterAtIndex:].cold.1 + 0
        3   CoreFoundation                      0x000000018a9ac97c -[__NSPlaceholderDictionary initWithCapacity:].cold.1 + 0
        4   CoreFoundation                      0x000000018a7dca60 -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 320
        5   CoreFoundation                      0x000000018a80e75c -[NSDictionary initWithDictionary:copyItems:] + 408
        6   CoreFoundation                      0x000000018a80e524 -[__NSPlaceholderDictionary initWithDictionary:copyItems:] + 172
        7   CoreText                            0x000000018be79678 _ZN11TAttributes9ApplyFontEPK14__CFDictionaryPK8__CTFontPK11TCharStream7CFRangePPKv + 252
        8   CoreText                            0x000000018be79430 _ZN11TAttributesC2EPK14__CFDictionaryPK11TCharStream7CFRange + 96
        9   CoreText                            0x000000018bdd4bc8 _ZN21TTypesetterAttrString10InitializeEPK20__CFAttributedString + 356
        10  CoreText                            0x000000018bdd48b8 _ZN21TTypesetterAttrStringC2EPK20__CFAttributedStringPK14__CFDictionary + 112
        11  CoreText                            0x000000018bdd4770 CTLineCreateWithAttributedString + 136
        12  UIFoundation                        0x000000018de060b4 -[NSCoreTypesetter _stringDrawingCoreTextEngineWithOriginalString:rect:padding:graphicsContext:forceClipping:attributes:stringDrawingOptions:drawingContext:stringDrawingInterface:] + 1808
        13  UIFoundation                        0x000000018dd4330c __NSStringDrawingEngine + 1964
        14  UIFoundation                        0x000000018dda78e0 -[NSAttributedString(NSExtendedStringDrawing) drawWithRect:options:context:] + 32
        15  ModuleKit                           0x00000001028248e8 block_destroy_helper.11 + 2344
        16  ModuleKit                           0x0000000102824468 block_destroy_helper.11 + 1192
        17  ModuleKit                           0x00000001028245a8 block_destroy_helper.11 + 1512
        18  AppKit                              0x000000018d1a6334 _NSViewDrawRect + 148
        19  AppKit                              0x000000018d8d7400 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1312
        20  AppKit                              0x000000018d1a5ab4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 676
        21  QuartzCore                          0x0000000190b4b908 -[CALayer renderInContext:] + 1316
        22  QuartzCore                          0x0000000190b4d258 -[CALayer _renderSublayersInContext:] + 412
        23  QuartzCore                          0x0000000190b4b91c -[CALayer renderInContext:] + 1336
        24  QuartzCore                          0x0000000190b4d258 -[CALayer _renderSublayersInContext:] + 412
        25  QuartzCore                          0x0000000190b4b91c -[CALayer renderInContext:] + 1336
        26  AppKit                              0x000000018d8d6948 -[NSView _renderLayerInContext:] + 288
        27  AppKit                              0x000000018d8d7180 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 672
        28  AppKit                              0x000000018d326380 -[NSView displayRectIgnoringOpacity:inContext:] + 820
        29  AppKit                              0x000000018d325fe8 __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 268
        30  AppKit                              0x000000018d325eb8 -[NSBitmapImageRep _captureDrawing:] + 80
        31  AppKit                              0x000000018d325e4c -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 176
        32  AppKit                              0x000000018d7ad5a4 -[NSStatusItem _updateReplicant:] + 1380
        33  AppKit                              0x000000018d2e9d64 -[NSStatusItem _updateReplicants] + 220
        34  AppKit                              0x000000018d7adf50 -[NSStatusItem _updateReplicantsUnlessMenuIsTracking:] + 128
        35  Foundation                          0x000000018b619fe0 __NSFireDelayedPerform + 444
        36  CoreFoundation                      0x000000018a86fdb4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
        37  CoreFoundation                      0x000000018a86f9b0 __CFRunLoopDoTimer + 1076
        38  CoreFoundation                      0x000000018a86f438 __CFRunLoopDoTimers + 392
        39  CoreFoundation                      0x000000018a854644 __CFRunLoopRun + 1944
        40  CoreFoundation                      0x000000018a853734 CFRunLoopRunSpecific + 600
        41  HIToolbox                           0x0000000192751b84 RunCurrentEventLoopInMode + 292
        42  HIToolbox                           0x00000001927518f8 ReceiveNextEventCommon + 552
        43  HIToolbox                           0x00000001927516b8 _BlockUntilNextEventMatchingListInModeWithFilter + 72
        44  AppKit                              0x000000018d03d4ec _DPSNextEvent + 836
        45  AppKit                              0x000000018d03be8c -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1292
        46  AppKit                              0x000000018d02dd18 -[NSApplication run] + 596
        47  Stats                               0x000000010268fca8 Stats + 15528
        48  libdyld.dylib                       0x000000018a774420 start + 4
)
libc++abi: terminating with uncaught exception of type NSException
[1]    27147 abort      /Applications/Stats.app/Contents/MacOS/Stats
 ✘ lanthean@mba ➤ ~/mnt ➤ uname -a
Darwin mba 20.4.0 Darwin Kernel Version 20.4.0: Fri Mar  5 01:14:02 PST 2021; root:xnu-7195.101.1~3/RELEASE_ARM64_T8101 arm64
 ✔ lanthean@mba ➤ ~/mnt ➤
```
Command that failed
```sh
/Applications/Stats.app/Contents/MacOS/Stats
```

exelban, 2021-05-03:
Hi. Thanks for reporting.
The previous version works fine?

exelban, 2021-05-03:
Could I ask to you run the app without a battery module?

lanthean, 2021-05-03:
Yes, previous version worked ok.
Sadly I am unable to install the older version now through brew.

BR,
martin

exelban, 2021-05-03:
You can download the previous version directly from [github](https://github.com/exelban/stats/releases/download/v2.5.11/Stats.dmg).

Strange, there was no changes which can cause this type of crash.

lanthean, 2021-05-04:
> Could I ask to you run the app without a battery module?
Sorry for the delay. Just trying. And it only lasts for a little longer before crashing.
Please find below the dump:
```sh
 ✔ lanthean@mba ➤ ~ ➤ /Applications/Stats.app/Contents/MacOS/Stats
Error call(READ_KEYINFO): (iokit/common) invalid argument
Error read(): (iokit/common) invalid argument
Error call(READ_KEYINFO): (iokit/common) invalid argument
Error read(): (iokit/common) invalid argument
2021-05-04 10:27:40.120 Stats[96474:7521062] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]'
*** First throw call stack:
(
        0   CoreFoundation                      0x000000018a8d3db8 __exceptionPreprocess + 240
        1   libobjc.A.dylib                     0x000000018a5fd0a8 objc_exception_throw + 60
        2   CoreFoundation                      0x000000018a99ebb4 -[__NSCFString characterAtIndex:].cold.1 + 0
        3   CoreFoundation                      0x000000018a9ac97c -[__NSPlaceholderDictionary initWithCapacity:].cold.1 + 0
        4   CoreFoundation                      0x000000018a7dca60 -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 320
        5   CoreFoundation                      0x000000018a80e75c -[NSDictionary initWithDictionary:copyItems:] + 408
        6   CoreFoundation                      0x000000018a80e524 -[__NSPlaceholderDictionary initWithDictionary:copyItems:] + 172
        7   CoreText                            0x000000018be79678 _ZN11TAttributes9ApplyFontEPK14__CFDictionaryPK8__CTFontPK11TCharStream7CFRangePPKv + 252
        8   CoreText                            0x000000018be79430 _ZN11TAttributesC2EPK14__CFDictionaryPK11TCharStream7CFRange + 96
        9   CoreText                            0x000000018bdd4bc8 _ZN21TTypesetterAttrString10InitializeEPK20__CFAttributedString + 356
        10  CoreText                            0x000000018bdd48b8 _ZN21TTypesetterAttrStringC2EPK20__CFAttributedStringPK14__CFDictionary + 112
        11  CoreText                            0x000000018bdd4770 CTLineCreateWithAttributedString + 136
        12  UIFoundation                        0x000000018de060b4 -[NSCoreTypesetter _stringDrawingCoreTextEngineWithOriginalString:rect:padding:graphicsContext:forceClipping:attributes:stringDrawingOptions:drawingContext:stringDrawingInterface:] + 1808
        13  UIFoundation                        0x000000018dd4330c __NSStringDrawingEngine + 1964
        14  UIFoundation                        0x000000018dda78e0 -[NSAttributedString(NSExtendedStringDrawing) drawWithRect:options:context:] + 32
        15  ModuleKit                           0x0000000102ee48e8 block_destroy_helper.11 + 2344
        16  ModuleKit                           0x0000000102ee4468 block_destroy_helper.11 + 1192
        17  ModuleKit                           0x0000000102ee45a8 block_destroy_helper.11 + 1512
        18  AppKit                              0x000000018d1a6334 _NSViewDrawRect + 148
        19  AppKit                              0x000000018d8d7400 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1312
        20  AppKit                              0x000000018d1a5ab4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 676
        21  QuartzCore                          0x0000000190b4b908 -[CALayer renderInContext:] + 1316
        22  QuartzCore                          0x0000000190b4d258 -[CALayer _renderSublayersInContext:] + 412
        23  QuartzCore                          0x0000000190b4b91c -[CALayer renderInContext:] + 1336
        24  QuartzCore                          0x0000000190b4d258 -[CALayer _renderSublayersInContext:] + 412
        25  QuartzCore                          0x0000000190b4b91c -[CALayer renderInContext:] + 1336
        26  AppKit                              0x000000018d8d6948 -[NSView _renderLayerInContext:] + 288
        27  AppKit                              0x000000018d8d7180 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 672
        28  AppKit                              0x000000018d326380 -[NSView displayRectIgnoringOpacity:inContext:] + 820
        29  AppKit                              0x000000018d325fe8 __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 268
        30  AppKit                              0x000000018d325eb8 -[NSBitmapImageRep _captureDrawing:] + 80
        31  AppKit                              0x000000018d325e4c -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 176
        32  AppKit                              0x000000018d7ad5a4 -[NSStatusItem _updateReplicant:] + 1380
        33  AppKit                              0x000000018d2e9d64 -[NSStatusItem _updateReplicants] + 220
        34  AppKit                              0x000000018d7adf50 -[NSStatusItem _updateReplicantsUnlessMenuIsTracking:] + 128
        35  Foundation                          0x000000018b619fe0 __NSFireDelayedPerform + 444
        36  CoreFoundation                      0x000000018a86fdb4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
        37  CoreFoundation                      0x000000018a86f9b0 __CFRunLoopDoTimer + 1076
        38  CoreFoundation                      0x000000018a86f438 __CFRunLoopDoTimers + 392
[... cut ...]
        45  AppKit                              0x000000018d03be8c -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1292
        46  AppKit                              0x000000018d02dd18 -[NSApplication run] + 596
        47  Stats                               0x0000000102e4fca8 Stats + 15528
        48  libdyld.dylib                       0x000000018a774420 start + 4
)
libc++abi: terminating with uncaught exception of type NSException
[1]    96474 abort      /Applications/Stats.app/Contents/MacOS/Stats

```

exelban, 2021-05-04:
So, it still crashes with a disabled battery module?

lanthean, 2021-05-04:
> So, it still crashes with a disabled battery module?

yes

exelban, 2021-05-04:
Hmm, does it crash on some action?
Maybe when opening some popup window?
Or settings?

Sorry that I trying to guess. But a crash report just says that there is some internal bug (system). In 99% it causes when trying to calculate the text width. But it does not provide the trace. So there is only one way to find the bug - trying to find a module that causes it. (I don't have m1)

altrdev, 2021-05-04:
I tried to reproduce on MBP M1 but I couldn't.
I disabled Battery module, but no issue occurred

mrchoke, 2021-05-04:
Confirmed.
I use macmini M1 it crashed too.

merken, 2021-05-05:
Double confirmed, M1 Macbook Air, same problem

merken, 2021-05-05:
Reverting back to 2.5.11 works.

lanthean, 2021-05-05:
Actually no action is needed (on my experience). The 2.5.12 starts and crashes after a little while (e.g. few seconds) - no user actions.
Reverting to 2.5.11 helped - I am having no issue with this version.

exelban, 2021-05-05:
Could someone check this build for the crash?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6426140/Stats.dmg.zip)

lanthean, 2021-05-05:
> Could someone check this build for the crash?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/6426140/Stats.dmg.zip)

Testing..

merken, 2021-05-05:
Sensors are not displayed in that last custom build: 
[image]

exelban, 2021-05-05:
Yes, I know.

lanthean, 2021-05-05:
Seems that the custom build is stable. What has changed?

exelban, 2021-05-05:
Disabled sensors module.

exelban, 2021-05-05:
At least I know in what module bug.

altrdev, 2021-05-05:
2.5.12 no issue on MacBook Pro, maybe some edge case (some different sensors maybe?) with Air and Mini?

exelban, 2021-05-05:
It related to drawing. It crashes because the string which was drawn a moment ago was cleaned by GC. It's the best what I can understand from the crash. I cannot reproduce this bug. It also happened on [intel](https://github.com/exelban/stats/issues/136). So it does not relate to silicon.

exelban, 2021-05-05:
Please try this one:
[Stats.dmg.zip](https://github.com/exelban/stats/files/6426536/Stats.dmg.zip)

merken, 2021-05-05:
Thanks, testing it now...

merken, 2021-05-05:
Seems to be running fine for the past hour or so...

exelban, 2021-05-05:
Hmm, very very interesting. Could I ask someone else who has crash to try the last build?

RajceP, 2021-05-05:
My crash stack after enabling sensors: 

```Process:               Stats [2750]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.5.12 (236)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [2750]
User ID:               501

Date/Time:             2021-05-05 15:26:08.704 +0200
OS Version:            macOS 11.3.1 (20E241)
Report Version:        12
Anonymous UUID:        96BF384E-0E9F-9D6D-F35E-1E0AA7D96F6E

Sleep/Wake UUID:       A416526C-6B3C-4091-820F-348BC1F2AA51

Time Awake Since Boot: 64000 seconds
Time Since Wake:       2100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x0000000199207db8 __exceptionPreprocess + 240
1   libobjc.A.dylib                     0x0000000198f310a8 objc_exception_throw + 60
2   CoreFoundation                      0x00000001992d2bb4 -[__NSCFString characterAtIndex:].cold.1 + 0
3   CoreFoundation                      0x00000001992e097c -[__NSPlaceholderDictionary initWithCapacity:].cold.1 + 0
4   CoreFoundation                      0x0000000199110a60 -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 320
5   CoreFoundation                      0x000000019914275c -[NSDictionary initWithDictionary:copyItems:] + 408
6   CoreFoundation                      0x0000000199142524 -[__NSPlaceholderDictionary initWithDictionary:copyItems:] + 172
7   CoreText                            0x000000019a7ad678 _ZN11TAttributes9ApplyFontEPK14__CFDictionaryPK8__CTFontPK11TCharStream7CFRangePPKv + 252
8   CoreText                            0x000000019a7ad430 _ZN11TAttributesC2EPK14__CFDictionaryPK11TCharStream7CFRange + 96
9   CoreText                            0x000000019a708bc8 _ZN21TTypesetterAttrString10InitializeEPK20__CFAttributedString + 356
10  CoreText                            0x000000019a7088b8 _ZN21TTypesetterAttrStringC2EPK20__CFAttributedStringPK14__CFDictionary + 112
11  CoreText                            0x000000019a708770 CTLineCreateWithAttributedString + 136
12  UIFoundation                        0x000000019c73a0b4 -[NSCoreTypesetter _stringDrawingCoreTextEngineWithOriginalString:rect:padding:graphicsContext:forceClipping:attributes:stringDrawingOptions:drawingContext:stringDrawingInterface:] + 1808
13  UIFoundation                        0x000000019c67730c __NSStringDrawingEngine + 1964
14  UIFoundation                        0x000000019c6db8e0 -[NSAttributedString(NSExtendedStringDrawing) drawWithRect:options:context:] + 32
15  ModuleKit                           0x00000001026d48e8 block_destroy_helper.11 + 2344
16  ModuleKit                           0x00000001026d4374 block_destroy_helper.11 + 948
17  ModuleKit                           0x00000001026d45a8 block_destroy_helper.11 + 1512
18  AppKit                              0x000000019bada334 _NSViewDrawRect + 148
19  AppKit                              0x000000019c20b400 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1312
20  AppKit                              0x000000019bad9ab4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 676
21  QuartzCore                          0x000000019f47f908 -[CALayer renderInContext:] + 1316
22  QuartzCore                          0x000000019f481258 -[CALayer _renderSublayersInContext:] + 412
23  QuartzCore                          0x000000019f47f91c -[CALayer renderInContext:] + 1336
24  QuartzCore                          0x000000019f481258 -[CALayer _renderSublayersInContext:] + 412
25  QuartzCore                          0x000000019f47f91c -[CALayer renderInContext:] + 1336
26  AppKit                              0x000000019c20a948 -[NSView _renderLayerInContext:] + 288
27  AppKit                              0x000000019c20b180 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 672
28  AppKit                              0x000000019bc5a380 -[NSView displayRectIgnoringOpacity:inContext:] + 820
29  AppKit                              0x000000019bc59fe8 __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 268
30  AppKit                              0x000000019bc59eb8 -[NSBitmapImageRep _captureDrawing:] + 80
31  AppKit                              0x000000019bc59e4c -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 176
32  AppKit                              0x000000019c0e15a4 -[NSStatusItem _updateReplicant:] + 1380
33  AppKit                              0x000000019bc1dd64 -[NSStatusItem _updateReplicants] + 220
34  AppKit                              0x000000019c0e1f50 -[NSStatusItem _updateReplicantsUnlessMenuIsTracking:] + 128
[... 529 lines cut ...]
MALLOC guard page                  288K       16 
MALLOC_MEDIUM (reserved)         960.0M        8         reserved VM address space (unallocated)
[... cut ...]
__OBJC_RO                         74.9M        1 
__OBJC_RW                         2576K        1 
__TEXT                           240.8M      374 
__UNICODE                          588K        1 
libnetwork                         128K        8 
mapped file                      225.3M       28 
shared memory                      912K       15 
===========                     =======  ======= 
TOTAL                              3.7G     2644 
TOTAL, minus reserved VM space     2.4G     2644 ```

exelban, 2021-05-05:
@RajceP please check the last posted build.

RajceP, 2021-05-05:
> @RajceP please check the last posted build.

No crashes so far.

exelban, 2021-05-05:
Just to ensure. Does the last build crash?

RajceP, 2021-05-05:
> Just to ensure. Does the last build crash?

It does't crash.

exelban, 2021-05-05:
Ok, so I can release a fix.
Thanks.

exelban, 2021-05-05:
I have released a [hotfix](https://github.com/exelban/stats/releases/tag/v2.5.13) for this bug.
Thanks everyone for help.

closed 2021-05-05 by exelban

btannous, 2021-05-05:
this hotfix version also fixes my crash on m1 + 2.5.12! Thank you!