exelban/stats #454: Fans control will not stay on

thereald3bug, 2021-05-03:
Every time Stat runs on boot, the fans control is gone regardless of the fact that I keep turning it on every single time.  There seems to be no way to keep it up there other than turning it on at every boot (again).

exelban, 2021-05-03:
Hi. Thanks for the proposal.
I think it could be a helpful feature.

exelban, 2021-05-04:
Hi. There is one problem with this feature. Fans control requires root. So, if I add the option to save fans setup. The app will require a password on every start. I don't really like behavior like this. But it looks like there is no other way.

thereald3bug, 2021-05-04:
I know some programs somehow keep that root access even after reboots (like Parallels for example).  How do they do it?  Maybe it requires setting permissions in system settings?

exelban, 2021-06-18:
Hi.
Could you please check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6678453/Stats.dmg.zip)

thereald3bug, 2021-06-19:
That version works perfectly!  It retains the manual setting after reboots, restarts, etc…. The only thing it doesn’t read properly is the fact that I have a UPS hooked up on my USB port (OS X sees this however).

Thank you

closed 2021-06-20 by exelban