exelban/stats #456: reorganize modules in more user-friendly way

lanthean, 2021-05-04:
Hello Exelban, is there a way to organize modules (I might have missed it)?
I found only one way to disable all modules and enable them one by one in the correct sequence.
It would be much better to be able to drag them to new locations, while in settings perhaps?

PS: The stats app is great - thanks!

exelban, 2021-05-04:
Hi. What do you mean by `drag them to new locations`?

lanthean, 2021-05-04:
Hi. I meant, when Stats are running (sitting in menu bar), it would be nice to be able to drag modules to interchange their location on the menu bar.
E.g. I've disabled and enabled battery module, which put it on the leftmost position. 
[image]
Dragging it to the right, would move it for example to the right of the temperature module.

exelban, 2021-05-04:
[Please use google.](https://www.macrumors.com/how-to/rearrange-macos-menu-bar-icons/)

closed 2021-05-04 by exelban

lanthean, 2021-05-04:
Ah, thank You. I assumed (wrongly), that Stats would be all-together (therefore by moving it, it would move all the modules).
The fact, that (dropbox, copyclip and "what is playing") were mingled seemed to me as a insignificant bug.
I am sorry to have taken up Your time with this.