exelban/stats #458: Stats 2.5.12 Crash on Macbook Pro Retina, 13-inch, Early 2015 running Mac OS 10.15.7 (19H1030)

mivens, 2021-05-05:
**Describe the bug**
Since upgrading to 2.5.12, Stats crashes a few minutes after being launched with no other steps apart from running it needed to reproduce.

**Details:**
 - Device: Macbook Pro Retina, 13-inch, Early 2015 
 - MacOS  10.15.7 
 - Application version 2.5.12 

Crashfile attached (all four crash files I have contain the same backtrace).

[Stats_2021-05-05-134007_UKC17PNCD3FVH3.crash.txt](https://github.com/exelban/stats/files/6427103/Stats_2021-05-05-134007_UKC17PNCD3FVH3.crash.txt)

exelban, 2021-05-05:
Hi. Thanks for reporting.
Do you have installed IPG (Intel power gadget)?

exelban, 2021-05-05:
Ok. It's not related to the IPG.
Could you please try the last build posted on #453?

mivens, 2021-05-05:
Thanks for the quick reply.
The build posted at [453 comment-832593350](https://github.com/exelban/stats/issues/453#issuecomment-832593350) seems to fix the issue - no crashes so far. 
Reverting back to 2.5.12 and it crashes again after a few mins.

exelban, 2021-05-05:
thanks. I will close this issue because it looks like the same bug.

closed 2021-05-05 by exelban