exelban/stats #461: Duplicate upload speed in the Network widget

richardtop, 2021-05-07:
**Describe the bug**
[image]

**Details:**
 - Device: Macbook Pro 16
 - MacOS 11.3.1
 - Application version 2.5.13

exelban, 2021-05-07:
Hi. Thanks for the feedback.
There was already some issue with the explanation of this bug. As I remember it's related to the interface. A better way will be to use a process-based reader, not an interface-based one.
It cannot be fixed from my side. (I was trying to find a full discussion about this, but it looks like it was in some not network related ticket).

closed 2021-05-07 by exelban

meiser79, 2021-05-08:
It was #296.

richardtop, 2021-05-08:
How about setting "process based" as the default render type?

exelban, 2021-05-08:
No, it cannot be the default reader.