exelban/stats #462: Suggestion not bug report...

thereald3bug, 2021-05-07:
I have a small suggestion about the fan graphic for the toolbar.  Add a pie chart style and/or bar style to indicate how fast the fan is going like the other charts.

exelban, 2021-05-07:
Hi. Thanks for the feedback.
Do you propose to show fan speed as 0-100% in the bar chart?

thereald3bug, 2021-05-07:
Exactly… also, I have a question going in the apple dev forums to see if anyone can tell me how to retain root after reboots/restarts… hopefully it gets some useful responses.

exelban, 2021-05-07:
I will think about this feature.

About #454. There is no way to run the app as root after the restart. But I have made some research already. And I know how it can be implemented. But it will take time.

thereald3bug, 2021-05-07:
Was I on the right track (a system preferences thing under security maybe?) ?

closed 2021-05-16 by exelban