exelban/stats #467: Missing strings in localization files

kennymc-c, 2021-05-09:
There are some strings in the app that are only shown in English and not appearing in the localization files.
For example all widget settings sidebar items and headlines, some list entries like "sec", "Disabled" in battery notifications and the F and U letters in the menu bar icons.
It would be nice to make them translatable as well to have a coherent localization.

closed 2021-05-09 by exelban

kennymc-c, 2021-05-22:
@exelban Can or will this not be fixed or why has this issue been closed without a comment?

exelban, 2021-05-23:
Hi. It was closed because:
- items in the sidebar will be only in English, always
- all symbols in the menu bar will be only in English, always
- there are some elements that could not be translated