exelban/stats #473: CPU/GPU Sensor Apple M1

4lbertojordan, 2021-05-13:
What is the right CPU/GPU temperature sensor in Mac M1?

Any idea?

Thank you in advance

[image]

**Details:**
 - Device: Mac mini 2020 Apple M1
 - MacOS 11.3.1
 - Application version 2.5.14

exelban, 2021-05-14:
Hi.
I have no idea what these sensors mean(

closed 2021-05-16 by exelban

vallieres, 2021-08-24:
I understand this is not an issue per say, but I too would like to know what temperature to look at! :P 
I have a MacBook Pro M1 (2020) and have no clue what is the temperature I should look at that is equivalent to a "CPU Temperature" (I know the SOC is CPU+GPU+Memory; but which of these 40 sensor temps is the right one?)

vallieres, 2021-08-24:
OK so I found this little utility that seems to average the eACC and pACC (efficiency, performance cores) as the CPU temp, and both GPU sensors for the GPU temp. Maybe that could be an option in Stats?
https://github.com/andreivdev/coolm1/blob/main/app/computeutils.js

exelban, 2021-08-24:
@vallieres Hi. You could propose the mapping for sensors keys you know. I will add them to the app.
Average for sensors groups maybe will be added in the feature.

exelban, 2021-08-30:
@vallieres Hi. I have made some tests with m1 sensors. Add an average value, and rename some sensors.
Could you maybe check this build (I don't have m1 to check it)?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7075532/Stats.dmg.zip)

vallieres, 2021-08-30:
Sure! I installed it but can't see any changes.  Same list of sensors for temps, no average, unless it's not visible?

exelban, 2021-08-30:
hmm, there must be an average at least. Thanks, I will recheck it.

exelban, 2021-08-30:
@vallieres could you please check this one?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7077728/Stats.dmg.zip)

vallieres, 2021-08-30:
Amazing!

[image]

ximmox, 2021-08-31:
This looks great and works great @exelban and sorry for duplicating this in #601, my bad.

Averaging the values should be close enough to have an impression of the "one and only" CPU-temp. All the sensors are so close together that it won't matter if you take the highest of the sensors readings or averaging them.

exelban, 2021-08-31:
@ximmox no problems. Sorry, I just do not have time to describe it on your ticket. I will add this feature this week. I still have a few issues to solve related to this.