exelban/stats #476: REQ: mono accent on pie chart

gingerbeardman, 2021-05-14:
**Describe the feature request**

I'd like mono accent on the pie chart available in RAM section.

**Details:**
 - Device: Mac mini 2018
 - MacOS 11.3.1
 - Application version 2.5.14

**Alternative**
iStat Menus, where I'm coming from, offers this.

exelban, 2021-05-14:
Hi. Thanks for the proposal.
I'm not sure if there is a sense to have a monochrome pie chart. Because it's showing only multiple values. And it could be done only with different shades of gray.

gingerbeardman, 2021-05-14:
I like to keep colours out of my menu bar, reserving them only for alerts/errors/notifications. But not for something that is always visible.

What do you think?

(in iStat Menus I use mono arc display)

exelban, 2021-05-16:
Hi. I don't really understand what do you propose.

exelban, 2021-05-30:
Hi. There are some proposals of how it can be implemented?

closed 2021-06-08 by exelban