exelban/stats #479: REQ: bolder font for text based items

gingerbeardman, 2021-05-14:
**Describe the feature request**

Bolder font option for items such as network I/O

**Details:**
 - Device: Mac mini (Late 2018)
 - MacOS 11.3.1
 - Application version 2.5.14

**Alternative**
iStat Menus, where I'm coming from, offers this.

exelban, 2021-05-14:
Hi. Thanks for the proposal.
I'm not sure if it could be a boulder text. I'm trying to use the max font size and weight which will be drawn normally on different screens (dpi and resolutions). But if you have an idea of how it can be realized, you can make a PR and we will see if it's acceptable)

gingerbeardman, 2021-05-14:
I'll have a look/think about this! Thanks

exelban, 2021-05-30:
Hi. There are some changes?

gingerbeardman, 2021-05-31:
Sorry no, I don't think I will have time.

exelban, 2021-05-31:
Ok, I'm going to close this issue. It could be reopened if someone will have an idea of how it could be done.

closed 2021-05-31 by exelban

gingerbeardman, 2021-05-31:
OK, somewhat disappointing.

gingerbeardman, 2024-10-02:
Here's a comparison of iStats menus vs Stats

[image]