exelban/stats #482: REQ : Counter since last unplugging

Drozt-md, 2021-05-14:
# enhancement

_Hello first all of thanks for your incredible app, and sorry for my english, gonna try to be the most explicit._

I searched on the stats settings, and even in mac os (and maybe i'm blind), I cannot find a setting to show you the time you use the battery since the last time you unplugged the charger. 
Activity monitor can show that but it works only for one session, so if you shutdown your mac, It's reset

I know that you can see the time remaining of your battery but it 's just a estimate time, it depend of a lot of parameters like luminosity/wifi/etc, and it can be great if you can have a toggle in the setting to show a timer since you unplugged the charger, and the timer reset when you plugged again your mac.

I dont konw if it's technically possible, but I was thinking about that today, and I decided to write this suggestion.

Thanks for reading. And thanks again for your work :).

exelban, 2021-05-15:
Hi. Please take a look at #450

Drozt-md, 2021-05-15:
Oh my bad, didn't saw. 
Thanks for the reply.

closed 2021-05-15 by Drozt-md