exelban/stats #483: Blank space of sensor and fans symbols too large

leachimus, 2021-05-15:
Since the last two or three updates, the white space in the icons is way too big. This looks very unattractive in the bar. Please reduce the spacing again. Thanks

[image]

exelban, 2021-05-15:
Hi. Thanks for the feedback.
It's because previously the size of these widgets was dynamic. And not it's fixed.
There was a several reason for this. And also was a few asks for fixed width (#348).

exelban, 2021-05-15:
Hmm, I have an idea how it can be solved.
I can add an option: fixed width. The widget width will be dynamic by default. Based on widget value. And this option will make the width static.

I'm not sure if it's will be a good solution. Because of: #453 and #136. These crashes were connected to this widget width at some way.

leachimus, 2021-05-16:
Thanks for the support. It is worth a try. I will test and report. 👍

closed 2021-05-23 by exelban

leachimus, 2021-05-23:
Work very fine until now, thaks for fix it!