exelban/stats #485: Stats crashes

wfrank21, 2021-05-17:
Stats crashes after about an hour of runtime.
[Stats_2021-05-17-091611.txt](https://github.com/exelban/stats/files/6491928/Stats_2021-05-17-091611.txt)

**Details:**
 - Device: MacBook Pro 2017, 13-inch, Four Thunderbolt 3 Ports
 - MacOS 11.3.1
 - Application version 2.5.15

exelban, 2021-05-17:
Hi. Thanks for reporting.
According to the crash report it something with core draw. Could you please say if it crashes all time or it was 1 time crash?

wfrank21, 2021-05-17:
It crashed two times since updating. Running for the third time for about 40mins right now.

exelban, 2021-05-17:
Strange, because there was no changes in this part of app from a year.

wfrank21, 2021-05-17:
I will wait for it to crash again, then try rebooting the system. Maybe it's something with my Mac.

wfrank21, 2021-05-18:
It crashed again, even after multiple reboots. 
[Stats_2021-05-18-082055.txt](https://github.com/exelban/stats/files/6499866/Stats_2021-05-18-082055.txt)

exelban, 2021-05-18:
Hi. Thanks for the crash. It's a different crash than yesterday. And I think I know of how this one could be fixed.
I will make a test build later today.

wfrank21, 2021-05-19:
I have disabled the disk module. Waiting now...

exelban, 2021-05-19:
I have noticed this bug on one of my macs too( Disabling the disk module helps me.

Today will try to fix it.

wfrank21, 2021-05-20:
You're right, since disabling the disk module, no crash.

exelban, 2021-05-20:
Could you please check this build?
It must work without crashing.

[Stats.dmg.zip](https://github.com/exelban/stats/files/6513735/Stats.dmg.zip)

wfrank21, 2021-05-20:
Seems to work now with Disk Module.

closed 2021-05-23 by exelban