exelban/stats #487: After restarting, order of items is changed

bartkummel, 2021-05-20:
**Describe the bug**
After restarting the app (or rebooting the system), the order in which the items appear in  the menu bar is unpredictable. This is especially a problem if you e.g. combine the separate "NET" label with the network graph. You need those to be side-by-side. But it would be nice to be able to set up a fixed order for all of the items.

This screenshot shows what I mean.
[image]

**Details:**
 - Device: MacBook Pro (16-inch, 2019)
 - MacOS: 10.15.7 
 - Application version: 2.5.15

exelban, 2021-05-20:
Hi. Thanks for the feedback.
The order of items is controlled by the system. I will check if the `label` widget has the correct id (macOS is keeping the order by ids).

bartkummel, 2021-05-20:
It seems to me the order is determined by the order in which the items are enabled. I can fix the order by disabling and enabling the items in the right order. When I enable them again, each item is appended to the left hand side of the icon area in the menu bar. However, this order is lost when the app restarts.

exelban, 2021-05-20:
And what if you order the items manually?

maiux, 2021-05-21:
On Big Sur 11.3.1 if you hold Cmd and drag the items on the menu bar, the order you choose will be kept also if you close and reopen Stats.

bartkummel, 2021-05-21:
How do I do that?

closed 2021-05-21 by exelban