exelban/stats #492: White background and font when updating stats

spartan195, 2021-05-24:
After some updates individual tiles goes blank until you click on it. 
White font and background, it draws a white box between other stats, if you really focus on it you can see the values changing by the font outline.

Looks like it only does this on white font tiles.

The ones with white fonts are the ones that goes blank, the other two are fine.

[image]

**Details:**

 - Device: [Macbook Pro 2017]
 - MacOS [10.15.7]
 - Application version [2.5.16]

exelban, 2021-05-24:
Hi. Thanks for the feedback.
Does it happen only with the sensors widget?
If yes, could you please switch to the `static width` in the sensor's settings?

closed 2021-06-08 by exelban

spartan195, 2021-06-10:
Sorry for not replying in time,

Actually I've been testing, and it does it when a second monitor is connected.

Then, the white background appears on the one you are using, if you have two windows on each scree, when you click one or the other, the white background follows to the focused screen.

It does it too with the battery widget sometimes

exelban, 2021-06-10:
Thanks for sharing. I will check this behavior.