exelban/stats #494: please add drive throughput

Stooovie, 2021-06-01:
Thanks for the great app!

Disk throughput (aka "Finder is reading 120 MB/s", "Premiere is writing 50 MB/s") would be extremely useful. AFAIK only iStatMenus provides that functionality.

exelban, 2021-06-01:
Hi. Thanks for the proposal.
I'm not sure if it's possible with native API. But I will take a look.
Another problem is how to visualize these stats. Do you have some propose?

Stooovie, 2021-06-01:
iStatMenus has it rather simple but working. Of course ideally there would be indication what drive is what process accessing (by popover probably?) but this is sufficient for identifying stuck copying, video editing bottlenecks, stuff like that. Thanks!

[image]

exelban, 2021-06-01:
I take a look at this. The only way to fetch stats (read/write) per process is `iotop`. But there is one big problem. It requires root.

Stooovie, 2021-06-01:
AFAIK tools like iStatMenus use a helper app to do that but that's probably too much work, don't worry about it :)

exelban, 2021-06-01:
Yeap, it looks like they use it.
I'm currently working on SMC module as a helper app. It's not super easy. And sorry, but I'm not ready to add and support another one only for this feature. 

To be honest, I'm not a big fan of apps which requires a root access. And if I will add this feature (even with a helper app), the app will be required `root` at the start. And I'm trying to avoid this.

closed 2021-06-01 by exelban