exelban/stats #496: Module langs translation?

ppizarror, 2021-06-03:
Hi, As my current software lang is Spanish I figured that the module's translation is not displayed; if I open Battery or Network the English lang is shown instead.

[image]

I think that the language should be consistent across the entire software, for instance, If I open the battery, the plugin's title should be "Batería" instead of "Battery", the same as the rest of the plugins:

[image]

I'm using the latest version 2.5.17.

PD: Sorry if it's the intended behavior.

exelban, 2021-06-04:
Hi. Thanks for the feedback. But as I say many times, these parts (module names) will be not translated. They will be always in English.

milotype, 2021-06-27:
@exelban It is really a pitty that you don't want to allow translations for module names!

I definitely aggree with @ppizarror ... it would certainly improve the quality of your app. I don't see any reason why they sholdn't be translated ...

I'm not a programmer, so I can't help to implement it ... but maybe there are other GitHub users who could help you out with it ...

exelban, 2021-06-28:
Hi. Ok, I will review this decision.

closed 2021-07-04 by exelban