exelban/stats #499: RAM, Top processes:

anddis, 2021-06-08:
The pop-up window with the Top processes doesn't display the RAM usage correctly if the usage is >1000 Mb.
For example, in the attached screenshot, Window Server is actually using 1,204 Mb of RAM and Outlook 1,143 Mb.

[image]

Thank you and all of the contributors for developing and maintaining Stat. Keep it up!

anddis, 2021-06-08:
Here you go. Thanks.

```
Andreas-MacBook-Pro-2:~ anddis$ top -l 1 -o mem -n 8 -stats pid,command,mem
Processes: 450 total, 2 running, 448 sleeping, 2346 threads 
2021/06/08 23:19:28
Load Avg: 1.94, 1.76, 1.68 
CPU usage: 1.90% user, 7.39% sys, 90.69% idle 
SharedLibs: 243M resident, 60M data, 39M linkedit.
MemRegions: 200920 total, 6412M resident, 161M private, 5020M shared.
PhysMem: 28G used (4209M wired), 3940M unused.
VM: 3188G vsize, 1377M framework vsize, 13171583(0) swapins, 13864436(0) swapouts.
Networks: packets: 11678833/20G in, 12414830/16G out.
Disks: 5947899/129G read, 6005798/127G written.

PID    COMMAND          MEM   
174    WindowServer     1442M+
31259  Microsoft Outloo 759M+ 
33578  Spotify Helper ( 581M+ 
24275  RStudio          564M+ 
7732   Google Chrome He 526M+ 
24294  rsession         518M+ 
329    Finder           488M+ 
7720   Google Chrome    437M+
```

exelban, 2021-06-08:
Oh, I think I know where is the problem. Just ignore the previous comment)

anddis, 2021-06-09:
Ok but FYI the `pid`s from the `top` command match the `pid`s from the Top process pop-up window (hover box)

closed 2021-06-13 by exelban