exelban/stats #500: Feature Request: All-In-One Menu Bar Icon

SteradianAdrian, 2021-06-09:
There could be an all-in-one menu bar icon option. Having separate icons for every stat can really fill up your menu bar.

Thinking about how Stats Panel does it:
[image]

exelban, 2021-06-18:
Hi. Thanks for the proposal.
Sorry for the long response, but I needed some time to think about it.

In theory, it could be implemented. But there are a few moments:
- when this view will be visible (on each widget, when all widgets disabled and when app icon left in the menu bar etc)
- new UI/UX
- which elements of each module must be presented there

For me, it looks like it must be a separate app. Because Stats actually has a different way of use than an app with a fuature like this.

closed 2021-07-04 by exelban

asyba, 2021-11-26:
> Hi. Thanks for the proposal. Sorry for the long response, but I needed some time to think about it.
>
> In theory, it could be implemented. But there are a few moments:
>
> * when this view will be [...]

so its going to be implemented or you are still thinking?

exelban, 2021-11-26:
@asyba I think it will be implemented someday.