exelban/stats #501: [MBP m1] stats process abnormally high memory consumption

Mr0cket, 2021-06-09:
**Describe the bug**
Noticed The stats process has very high memory usage; It was highest memory process on my system (while I had ~10 tabs of chrome!). Refer to the following screenshot for details:

[image]

**Details:**
 - Device: Macbook Pro (M1 2020)
 - MacOS: 11.2.3
 - Application version: 2.5.10

exelban, 2021-06-09:
Hi. Thanks for reporting.
It's totally not ok. Looks like some memory leak. Does it always was ram usage like this, or it's started with some update (system or Stats)?

Mr0cket, 2021-06-09:
Ahh good to know! 
this is the first time I have noticed the stats process with so much ram usage. but I just restarted my laptop, so will continue to monitor in case it occurs again.

exelban, 2021-06-09:
Please give me know if memory consumption will be more than 100Mb. It will mean that there is for 100% memory leak.

Mr0cket, 2021-06-14:
checked memory usage today, stats is again consuming ~1GB RAM.

[image]

gabormagyar, 2021-06-15:
Can confirm, had this instance running for a couple of days now. Leaking like crazy.

[image]

Have now disabled sensors and restarted Stats, will see if the leak comes back.

Mr0cket, 2021-06-15:
I have also disabled everything but RAM & GPU for the past 24 hrs. Will report back if process RAM increases again

gabormagyar, 2021-06-15:
I checked my version and I was on version 2.5.13. From what I can tell the fix to #417 was released in 2.5.14. (You seem to be on 2.5.10. according to the initial issue message.)

I checked out master and ran Stats in XCode and didn't see any leak, with or without sensors module.

Mr0cket, 2021-06-15:
Good catch, just upgraded to 2.5.19 & RAM usage hasn't spiked past 60 for me either! 
Closing this issue as the problem is resolved.

Mr0cket, 2021-06-15:
Good catch, just upgraded to 2.5.19 & RAM usage hasn't spiked past 60 for me either! 
Closing this issue as the problem is resolved.

Thanks @exelban.

closed 2021-06-15 by Mr0cket