exelban/stats #507: Show GPU type not working

1546545, 2021-06-16:
I have a 16inch macbook pro and i turn on the show GPU typeand it doesn't change in the menu bar. It worked in older versions but it does not work now.
[image]

exelban, 2021-06-17:
Hi. Thanks for the feedback.
Could you please give a little bit more details? Pleases share the popup view screenshot of the GPU module.

1546545, 2021-06-17:
when i change the setting called show GPU type it does not change anything that i can see. here is the image of the drop down.
[image]

exelban, 2021-06-18:
According to the screenshot, everything is fine. Could you please tell me a little bit more about what does not work?

exelban, 2021-06-18:
Ohh, ok. I see where is the problem. I will take a look at why it does not work properly.

closed 2021-06-20 by exelban