exelban/stats #511: Sensors show mixed scale

thereald3bug, 2021-06-19:
Sensors show both C and F scales.   Ambient is in C and GPU Proximity is in F.  There also seems to be no way to switch to either scale.

exelban, 2021-06-20:
Hi. Could you provide a screenshot of the sensors with the wrong scales?

thereald3bug, 2021-06-20:
[cid:560A337B-CA4D-4185-88AF-02901C20FC88]

Even though Ambient indicates F on the popup, the temp must be in C as 55F is impossible in this room.
It is impossible to select the scale to use however as there is no option apparently.  The Ambient sensor is attached to the CPU fan which should be the highest sensor usually.
If you convert 55C to F, that is 131 degrees which makes sense… however it indicates that it is in F (it is C)… and yet the others are correctly displaying F scale.

Also, I don’t know if this is normal or not, but Voltages for GPU, Memory and CPU Core 1 show 0.00

exelban, 2021-06-21:
Hi. I'm not sure I understand correctly. Do some sensors show as `F` and some as `C` at the same time?

thereald3bug, 2021-06-21:
It seems the problem lies with the Ambient sensor specifically.  It’s indicating that it’s showing you the Fahrenheit scale when the number is actually Celsius.  I’ll explain why it is incorrect.  As shown in the picture, it shows 55 degrees which is impossible.  My room temperature is 77 degrees, so unless there is a refrigeration system in this iMac then that temp is actually in Celsius, but the software is indicating that it’s in Fahrenheit which would be 131 degrees Fahrenheit… which is the number it should be displaying.

Additionally, you may want to look into the power sensors… it is not reading “GPU”, “Memory” or “CPU core 1” power.  There is a utility that is open source that displays these correctly… maybe you could have a look and see what is wrong there:  HWSensors. https://sourceforge.net/projects/hwsensors/

exelban, 2021-06-21:
Please, could you provide some screenshots?
There are a lot of sensors providing by SMC which has wrong values. At least there are a lot of negative values. 
All sensors use the same function for conversion. It will use system units if you do not change the default temperature unit in the settings. And I have no idea of how it's possible to see `C` and `F` at the same time.

If you know that some sensors are missing and you know the key and name of sensors, you could complete the [list of known sensors](https://github.com/exelban/stats/blob/dcc996d0bd5fb6e27a207d36effaffe2aeb2afe4/Modules/Sensors/values.swift#L105-L181) with PR.

thereald3bug, 2021-06-22:
I am not sure which of these need to be added to list of known sensors, but here is what they are using:

exelban, 2021-06-27:
Hi. Sorry but I do not understand the problem. If the values show in both scales, please provide a screenshot. Because I'm pretty sure that it's not possible. If you want to add some sensors to the list, you could open a separate issue with keys that need to be added (or PR will be better).
Thanks.

closed 2021-06-27 by exelban