exelban/stats #513: Ability to Set Fan Curves Based on Temperature

SidSidSid16, 2021-06-21:
Would it be possible to allow the user to set fan curves based on CPU, GPU and battery temperature readings? I'm not too sure about the technicalities but I think it should be possible as the CPU, GPU and battery stats widgets do have a temperature reading and the stats app has the means to set fan RPM.

exelban, 2021-06-21:
Hi. Thanks for the proposal. 

I'm pretty sure that it will not be added to the app.
I have added fan control because there was already monitoring. And simple control does not require complicated UI. 
There are nice apps that allow control fans with complicated flows: [smcFanControl](https://github.com/hholtmann/smcFanControl), [MacsFanControl](https://crystalidea.com/macs-fan-control).

Why it will not be added:
- requires complicated UI/UX
- requires added logic of one (Sensors) module to another (Fans)
- provides a lot of places where potentially could be a lot of bugs (at this moment fan control does not work on macs with T2)
- Stats is more about monitoring, not control

closed 2021-06-21 by exelban

holdit, 2021-11-29:
I completely understand why this isn't being added and _exelban_ gave good reasons above, but with the new Macs, I feel the options available are not good enough.

The main "problem" is that M1 Macs usually don't heat up enough to require fans to be running all the time. All you need is a more aggressive curve after a certain temperature.

- MacsFanControl: Works with M1, but it has a limitation... it keeps the fans running at their lower speeds even when they could be off. An issue was opened back in 2018, but hasn't been fixed yet: https://github.com/crystalidea/macs-fan-control/issues/1

- iStat Menus: Works with M1 and they do seem to turn off fans when not needed, but just like on older Intel Macs, it sometimes doesn't change the fan speed or takes too long to do it. It has been buggy for years and it probably doesn't make sense to run _Stats_ and _iStat Menus_.

- smcFanControl: Only works on older Intel Macs. I haven't used it in a while it also didn't support more advanced uses, if I remember correctly.

So yeah. Alternatives exist, but it wouldn't hurt to have something a bit more "polished".