exelban/stats #515: CPU frequency not shown, altough Intel Power Gadget is installed (screenshot)

MartinV96, 2021-06-23:
I installed the Intel Power Gadget to check my CPU frequency and surprisingly noticed that a new option in Stats showed up in the CPU tab. ("CPU clock frequency (IGP)"). Altough I checked the box, there is no frequency showing up in the menu bar.

[image]

exelban, 2021-06-23:
Hi. Frequency is showing in the popup, not in the menu bar.
[image]

closed 2021-06-23 by exelban

Marvo96, 2021-06-23:
Oh, you're right, thank you. Are you planning to implement it into the menu bar? It would be really helpful :)
Thank you for your work!

exelban, 2021-06-23:
No, CPU frequency will be only in the popup.