exelban/stats #516: Feature Request: Add ping to network module

gallolu, 2021-06-24:
I would find the display of ping times in the network module useful. You would have to configure an IP/domain that is pinged regularly. If necessary, you could also define the frequency. 

I would find this useful because my network connection sometimes has problems and this can be seen well via the ping. For instance, if you're on a train and connected to a hotspot, you can't tell a bad connection by the wifi strength.

The ping times could be displayed as the current ping or the average of the last 5-10 pings. A chart may also be useful.

exelban, 2021-06-25:
Hi. Thanks for the proposal.
I was thinking a few times about this feature. And there are a few moments:
- it will be only in the popup view
- it will be a little bit slow (because it will start sending ping when you open the popup, and it will need some time to calculate an average ping)
- it uses a lot of CPU to make ping/pong
- chart will be not available, it will require running ping/pong when app running (see previous point why I don't like it)

gallolu, 2021-06-27:
Hi, thanks for considering this, I really appreciate that 😄 

Well actually, it could be more useful if it would be running all the time. So like the temps or network speed, that I can see constantly in the menubar, the ping should be always visible. 

I can understand the downside with the CPU load. Users could say "I'm aware of this and accept this" but this would mean

- a) there has to be some kind of "notice"
- b) you need to deal with unsatisfied users that bug you about "the tool using too much CPU 🤷"

Or there could be some kind of "expert mode" for all sorts of settings, that come with those downsides. Maybe there's more? 🤔 

Anyways: I see that it's not as easy as displaying another temperature, that's already offered by the OS 😄

exelban, 2021-06-30:
Hi.
No, it cannot be running all the time. It will take too many resources for such a small feature. I suppose will be better to have a separate app for something like this. Because this feature requires too many settings:
- how many pings take for the average ping
- what server use for ping (8.8.8.8, 1.1.1.1, etc)
- notifications about high ping
and more)

In theory, it could be realized as a separate module, but I will prefer to create a separate app.

closed 2021-07-04 by exelban