exelban/stats #517: Menu bar items consistency (alignment + width)

cristiean, 2021-06-25:
[image]

**Width**
Take, for example, the CPU item. It is unnecessarily wide on the right side. It seems like its width is larger than the other items'. The RAM item is also wider on the right side than it is on the left.

Would it be possible to shave some pixels off the sides so they look neater and tighter together (without looking too tight, of course)?

**Alignment**
Wouldn't a **centred alignment** make more sense for all menu bar items? This would make any combination of items look more consistent.

The Sensors item, for example, is right-aligned while the RAM and CPU items are left-aligned. People will want to mix and match menu bar items as they please.  If some items use left-alignment and other use right-alignment, it may lead to big gaps between them and an inconsistent look.

**Details:**
 - Device: [Macbook Pro 13" 2018]
 - MacOS [11.4]
 - Application version [2.5.20]

exelban, 2021-06-25:
Hi. There is somewhere a ticket where I was described why it's not possible. Ping me if you will not find it. I will describe it again)

cristiean, 2021-06-25:
As far as I understand, the width is dynamic. But what about the alignment?

exelban, 2021-06-25:
Nope, the width is static. The dynamic width will provide not nice bug with a jumping menu bar.

I choose the left side alignment for this widget because it looks nicer for me than the right side. The centered value does not look nice at all in the menu bar.

But I think it could be realized as an option in the settings for this widget. Where you can choose an alignment: left (by default), center, or right.

cristiean, 2021-06-25:
That would be a nice option

closed 2021-07-04 by exelban