exelban/stats #520: [Feature request] Maximum setting for fans

igorkulman, 2021-06-29:
The fans settings now have an "Automatic" and "Manual options

[image]

I would like a third option called something like "Maximum" that would run the fan at full speed. Mac Fans Control app calls it "Full blast". 

The use case for this is when you want to quickly cool down the machine.

**Details:**
 - Device: Macbook Pro 16"
 - MacOS 11.4
 - Application version 2.5.21

luckman212, 2021-07-12:
Related question @exelban : is it possible to set only a higher `minimum` value, but leave the fan mode in `auto` so it can fluctuate up and down? E.g. on my Mac Mini 2018, default is 1700 min / 4400 max. But this runs pretty hot, and even at 3500RPM I still cannot hear the fan. So I would like to set 3500 as the minimum but allow it to go up to 4400 if needed when it gets very hot. Is this possible?

exelban, 2021-07-12:
@luckman212 not sure. Setting the fans speed process contains 2 steps: change fan mode and set the target value.
So the fan could only in one mode: auto or manual. Not both at the same time. When the fans in auto mode, the target value is controlled by the system.

In theory, maybe, you could rewrite minimal value (there is a separate key for this) for the specific fan in the SMC. But it's not safe.

luckman212, 2021-07-12:
Ok, yes I tried this, but it does not "stick" (reverts back to 1700).  So, not sure...

```
mini:~ $ smc fans
Number of fans: 1.0

0: Fan #0
Actual speed: 1700.0218505859375
Minimal speed: 1700.0
Maximum speed: 4400.0
Target speed: 1700.0
Mode: automatic

mini:~ $ smc set -k F0Mn -v 3500
[INFO]: set 3500 on F0Mn

mini:~ $ smc list -f
[INFO]: found 22 keys

[F0AB]     229.0
[F0AH]     5000.0
[F0AL]     4400.0
[F0AP]     75.0
[F0Ac]     1702.2161865234375
[F0Dc]     0.24448297917842865
[F0Fb]     1.0
[F0Fc]     3.0
[F0Md]     0.0
[F0Mn]     1700.0
[F0Mx]     4400.0
[F0Sf]     0.0
[F0St]     5.0
[F0Tg]     1700.0
[FAC0]     0.0
[FAC1]     0.0
[FAC2]     0.0
[FAC3]     0.0
[FBAD]     0.0
[FNum]     1.0
[FOff]     0.0
[FRmp]     0.0
```

closed 2021-07-18 by exelban

bob-ortiz, 2021-07-19:
How do I turn off turbo boost? I tried it and it works perfectly, but after they are at maximum speed I cannot turn the fans back to normal automatic speed. To turn it off I had to set the fans to manual, then choose a random speed (minimum, for example) and then switch to "Automatic" again. Isn't there a way to just shut off "turbo boost" directly? I got pretty scared during those minutes the fans where at full speed.

exelban, 2021-07-19:
@bob-ortiz Hi. The simplest way is to switch to the manual mode. Manual mode automatically sets the proper fans speed.