exelban/stats #521: Feature Request: CPU Temperature

azeezsker, 2021-06-29:
Thanks for this beatiful little app, appreciate if you could include CPU temperature.
Regards

exelban, 2021-06-30:
Hi.
You could see the CPU temperature in the CPU popup, or enable it in the Sensors module.
[image]

ximmox, 2021-06-30:
I came looking for another feature suggestion, but it fits probably in here. I have a Macbook Air M1 (base model, 8GB RAM, 256GB SSD)

1. I can't see the temperature in the CPU-popup, without your screenshot I wouldn't have known that's possible.

2. feature suggestion: 
There are 9 sensors for my M1 CPU-temperature, 2x "eACC MTR Temp Sensor" and 7x "pACC MTR Temp Sensor", each with different numbers for the different cores (I suppose).

The problem: To see the highest CPU temperature, I would have to add all 9 sensors to the menubar or otherwise have to guess which i.e. 2 cores might run the hottest of them.
The little tool Hot by Macmade (https://github.com/macmade/Hot) solves that beautifully by only showing the hottest of the 9 sensors in the menubar, which is exactly what most of the users want out of cpu temp monitoring.

Maybe it could be possible to add a "virtual sensor" for the Sensors module in the menubar, which displays only the highest value of all the cpu sensors?

Otherwise I will keep using Hot alongside Stats, but I'd love to minimize the number of tools.

exelban, 2021-06-30:
@ximmox so the `eACC` and `pACC` are CPU temperature sensors?
Do you maybe know the difference between both of them?

ximmox, 2021-06-30:
Hi @exelban, I can't find a proper documentation right now, I was looking for a better description of the M1 sensors some weeks ago myself...

Here's a hint, but also without source: https://forums.macrumors.com/threads/istat-menus-now-sort-of-supports-temperature-sensors-on-m1.2276062/?post=29410903#post-29410903

> It seems pACC means performance cores, eACC means efficiency cores
However the number of sensors does not directly match with the 4+4 core layout of the M1 chips (mine is a 7 graphic core Air even.)

exelban, 2021-06-30:
@azeezsker do you have a mac with apple silicon?
@ximmox maybe it's a wrong place to discuss m1 sensors if this issue is related to the intel based. I'm interested to create a proper mapping for m1 sensors.

azeezsker, 2021-06-30:
@exelban: I am on a skylake hackintosh.

exelban, 2021-06-30:
Please ensure that your SMC (FakeSMC or VirtualSMC) provides one of these sensors: 
- TC0D
- TC0E
- TC0F
- TC0P
- TC0H

exelban, 2021-07-04:
Hi. Any updates?

azeezsker, 2021-07-05:
Hi Exelban, I was able to get the CPU Temperature as shown in your screen shot reply. Thanks.

closed 2021-07-05 by exelban