exelban/stats #523: Incorrect RAM Size denoted for a third party application - Notion

Atharva-Parikh, 2021-07-01:
**Describe the bug**
RAM memory utilisation for an application called Notion displaying incorrectly. The amount is correct but the size denoted is incorrect (I believe it should be in MB not GB as my total RAM capacity is 8 GB 😅.

**Details:**
 - Device: MacBook Air 13" (M1, 2020)
 - macOS Big Sur (11.4.0)
 - Application Version 2.5.21
  
[image]

exelban, 2021-07-01:
Hi. Thanks for your feedback.
Could you please share the output of `top -l 1 -o mem -n 8 -stats pid,command,mem`?

Atharva-Parikh, 2021-07-03:
Hey,
Here's the output of command:

> **top -l 1 -o mem -n 8 -stats pid,command,mem**

Hope it helps 😃

[image]

closed 2021-07-04 by exelban