exelban/stats #525: After the update, all usage rates show 1%

ReekyStive, 2021-07-04:
**Describe the bug**
After the latest update, all usage rates show 1%.
This issue is on [v2.5.22](https://github.com/exelban/stats/releases/tag/v2.5.21), while [v2.5.21](https://github.com/exelban/stats/releases/tag/v2.5.21) is working fine.
[image]

**Details:**
 - Device: Macbook Air 2020 (M1)
 - macOS 11.4 (20F71)
 - Application version 2.5.22

exelban, 2021-07-04:
Hi. Thanks for reporting. Will be fixed in a moment.

ReekyStive, 2021-07-04:
Oh! I was editing that file just now, ready to open a PR. Thank you for your repair!

ReekyStive, 2021-07-04:
Fixed in commit https://github.com/exelban/stats/commit/08ed4379f26352c791fd63b9a8afad8acc3a3a09.

closed 2021-07-04 by ReekyStive

exelban, 2021-07-04:
I updated a [release](https://github.com/exelban/stats/releases/tag/v2.5.22), please reinstall the app manually.

ReekyStive, 2021-07-04:
it's fixed, thanks!