exelban/stats #527: CPU, RAM 1%

cristiean, 2021-07-04:
CPU and RAM 1% after update

To note, I installed Bartender 4 today and am using their width-reduction feature, bringing all of the menu bar items closer together (as can be seen in the image). Not sure whether that impacts Stats in any way. When I quit Bartender, they're still on 1%.

 I have also used the new alignment features in Stats.

**Details:**
 - Device: [e.g. Macbook Pro 2018]
 - MacOS [e.g. 11.4]
 - Application version [e.g. 2.5.22]
[image]

ysomad, 2021-07-05:
Same
[image]

exelban, 2021-07-05:
Duplicate of #525

closed 2021-07-05 by exelban