exelban/stats #536: A lot of CPU by nettop

linkinek, 2021-07-11:
**Describe the bug**
I have a lot of CPU load by nettop process. It it happening after update to latest version and happens in randomly time.
CPU normalise only after close Stats and kill nettop process. And if I kill nettop process without close application, process starts again with a lot of CPU load.

**Details:**
 - Device: Macbook Pro 2018
 - MacOS 11.4 (20F71)
 - Application version 2.5.22

exelban, 2021-07-11:
Hi. Thanks for reporting.
Nettop is using to fetch top processes. And it must be running only when you open the popup view. Could you please run the Stats and do not click on the items in the menu bar at all. After some period of time check in the Activity Monitor if the problem appears.

linkinek, 2021-07-12:
Hi. 
I updated to 2.5.60 and problem exists yet. Nettop starts immediately after starting the application. But now there are 2 of them.
Can I attach some information for analyse?

exelban, 2021-07-12:
Hi. What type of network reader do you use?

linkinek, 2021-07-12:
What do you mean?

exelban, 2021-07-12:
There is an option `Reader type` in the network module. What option you have selected?

linkinek, 2021-07-12:
Process based

exelban, 2021-07-12:
Oh, now I see. Unfortunately, it cannot be fixed. 
If you have selected `Process based`, the app calculates network traffic based on `nettop` data. I could only recommend switching to the `Interface based`.

`nettop` is an internal macOS tool. And it's only one way to fetch network traffic without root permissions.

linkinek, 2021-07-12:
Oh... Is there a difference between Process based and Interface based?

exelban, 2021-07-12:
Yes. `Process based` using `nettop` and calculating network bandwidth based on the sum of all processes.
`Interface based` listening to the bandwidth on the specific interface which is active.

linkinek, 2021-07-12:
Ok. Thanks. I think can close the issue.

closed 2021-07-12 by exelban