exelban/stats #540: Integrate CPU throttling status display from "Hot" project?

luckman212, 2021-07-12:
I found this cool project on GitHub called 🔥[Hot](https://github.com/macmade/Hot) which shows CPU temp and throttling effects due to overheating. Since it's also written in Swift @exelban, I wonder if you'd consider adding some of this functionality into Stats. It would be nice if the CPU module e.g. could display this info as to whether the CPU was being throttled.

[image]

exelban, 2021-07-12:
Hi. Thanks for the feedback.
There was already a ticket (#193) with this proposal.

Could you please explain why you need to see this value?

luckman212, 2021-07-12:
Ah, sorry! I didn't see it was a duplicate.

After upgrading RAM in my Mac mini to 64GB, I noticed CPU was running extremely hot (~96C) and I was worried about throttling. Just thought it could be useful to display this status -- apparently there are a couple of other people who wanted to see this as well. But I understand it's a rare case and not very urgent.

exelban, 2021-07-12:
In my opinion, throttling could be noticed only by frequency.
I have played with the values `CPU_Scheduler_Limit` and `CPU_Speed_Limit`. And when my CPU definitely throttle (CPU drops frequency) these values were 100%. Maybe it's only my case. Or maybe intel turbo boots drop the frequency.

Could you confirm that these values (`pmset -g therm`) report the throttling?

luckman212, 2021-07-12:
Hmm, I'm not sure I have actually throttled either, 

```
$ pmset -g therm
Note: No thermal warning level has been recorded
Note: No performance warning level has been recorded
2021-07-12 17:23:05 -0400 CPU Power notify
	CPU_Scheduler_Limit 	= 100
	CPU_Available_CPUs 	= 12
	CPU_Speed_Limit 	= 100
```

But I will pay attention next time CPU gets very hot, and check again.

exelban, 2021-07-18:
Hi. Any updates?

luckman212, 2021-07-19:
Hi @exelban - thanks for asking.

Ok, I had a little time to play around. I found out `pmset -g therm` is not a good way to tell if throttling has happened. It's more of a real-time snapshot of conditions at that exact moment, so it usually will report nothing.

However, `pmset -g thermlog` does provide good info since it runs continously. I recorded this video, showing throttling happening as CPU heats up (using [stress-ng](https://kernel.ubuntu.com/~cking/stress-ng/) -- `brew install stress-ng`), this is easily reproducible for me now. So, yes I do think it would be nice/useful to display this in the Stats!

```shell
stress-ng --cpu 6 --io 4 --vm 2 --vm-bytes 128M --fork 4 --timeout 30s
```

https://user-images.githubusercontent.com/1992842/126157750-6e1b5ac2-47ff-4cbf-9f8f-c1456b52a0b3.mp4

closed 2021-08-29 by exelban

luckman212, 2021-08-29:
@exelban thanks for adding this! 👍

bob-ortiz, 2021-08-30:
How does this new feature work?

exelban, 2021-08-30:
@bob-ortiz in the CPU details you can see the Speed and Scheduler limits.