exelban/stats #542: FEATURE REQUEST: Per-day Internet usage stat

im-AMS, 2021-07-13:
1stly thanks for the amazing app!

Could y'all add sort of a toggle switch feature where in the Network widget shows internet usage per-day, instead of just showing internet used.

exelban, 2021-07-13:
Hi. Thanks for the proposal.
What do you mean, to show total usage over network i/o in the menu bar?

im-AMS, 2021-07-13:
I mean, we get lots of graph and data when we click on the net-speed widget right?

[image]

Like the highlighted above.

**Current Behaviour** 
The current feature just shows the internet usage since boot(im on mac Air), and since no one never shuts down a mac, it just keeps incrementing based on internet usage. One needs to press the reset  button to make Total Upload and Download 0GB

**Suggested Behaviour**
Make a toggle switch where the behaviour becomes, the Total Upload and Total Download automatically resets to 0GB daily at 12:00 AM.
If the toggle switch is disabled it stays the same behaviour as before.

exelban, 2021-07-13:
Ok, I understand. But it cannot be a toggle for sure)
I will try to add this feature.

im-AMS, 2021-07-13:
Thanks for accepting the request, really appreciate it!

closed 2021-09-19 by exelban