exelban/stats #543: Feature Improvement: Bluetooth

RoyRao2333, 2021-07-14:
The new Bluetooth feature is really useful, but it'll be better if not just choose which device you wanna present, but can present all bluetooth devices information like Battery section does (with a tiny popover). 

Furthermore, please make icons for disconnected devices like single AirPods earphone in case to display something like below:

[image]

Rather than:

[image]

This will be more clearer for users' perception.

__Again, thanks for making this app more and more powerful. :)__

---

MacBook Pro 2017, macOS 11.4

exelban, 2021-07-14:
Hi. Thanks for the feedback.
I'm not sure if I understand correctly.

If we are talking about elements in the menu bar. There is a battery widget that could show only 1 value (battery). 
I have a plan to add more widgets for this module. One is from the sensors module to show a few values at the same time. And maybe one dedicated to the AirPods (not sure, but I want to have this one).

In the popup view, you could see only connected devices to the mac. If the device is connected, it does not mean that the battery level is available. AirPods works in a little bit different way. MacOS handling AirPods battery level internally. So AirPods battery levels are available almost always.

I'm not sure what do you mean by these icons.

RoyRao2333, 2021-07-21:
> Hi. Thanks for the feedback.
> I'm not sure if I understand correctly.
>
> If we are talking about elements in the menu bar. There is a battery widget that could show only 1 value (battery).
> I have a [...]

Sorry for the mess... Let me explain more specifically.

I'd like the popover of Bluetooth section to show all the bluetooth devices currently connected to my Mac, rather than just selected ones. 😉

RoyRao2333, 2021-07-21:
> Hi. Thanks for the feedback.
> I'm not sure if I understand correctly.
>
> If we are talking about elements in the menu bar. There is a battery widget that could show only 1 value (battery).
> I have a [...]

Sorry for the mess... I get your point now, but I'm still confused about one thing:

When I wear both of my AirPods earphones, still can't get battery percentage from all of them as they're always 0%:

[image]

exelban, 2021-07-21:
Please take a look at #551

closed 2021-07-21 by exelban