exelban/stats #544: Battery percentage is wrong

pkumsy, 2021-07-14:
The battery percentage is not right. As can be seen, the system battery percentage is 100%, but stats is 97%...

[image]

 - Device: [Macbook Pro 2020 M1]
 - MacOS [11.4]
 - Application version [2.6.0]

exelban, 2021-07-14:
Hi. Thanks for reporting.
Could you please tell if it was some glitch and Stats battery level updates to the same value as the system?
Or it's for all time different than the system?

pkumsy, 2021-07-14:
It is always different. The battery percentage in stats was 99% two days ago, then 98% and now 97%. But it is always recharged these days.

exelban, 2021-07-14:
Hmm, it's strange. What if you restart the app?

pkumsy, 2021-07-14:
Ok...something more strange happened....
I restarted the app but it was still 97%. Then I restart the computer, before stats lunches, the system battery also shows 97%!! However, after several seconds the system battery becomes 100%, but stats battery keeps 97%.
Do you have any idea what is going on ...? Thanks.

pkumsy, 2021-07-14:
Ok, maybe I find the solution. After I turned off the **_optimized battery charging_**, the system battery becomes 97% again...

exelban, 2021-07-14:
Hmm, it makes sense.
Maybe it takes battery on 97%, but the system widget knows that `optimized battery charging` is enabled. So it's showing 100%.

closed 2021-07-14 by pkumsy