exelban/stats #547: Feature Request - Low Power Energy Mode

vysecurity, 2021-07-19:
Hi there, I have a small feature request under the battery bar.

Maybe it would be good to show if the low power energy mode is enabled, when you drop down the menu.

exelban, 2021-07-19:
Hi. Thanks for the proposal.
As I know this feature is available only on macOS 12+. And it could be added when it will be released.

closed 2021-07-19 by exelban

Atemu, 2022-03-01:
macOS 12 has been out for a while, could this be implemented?

Is it perhaps also possible to toggle it?

Atemu, 2022-03-01:
Read access is trivial:

https://developer.apple.com/documentation/foundation/nsprocessinfo/1617047-lowpowermodeenabled  
https://www.hackingwithswift.com/example-code/system/how-to-detect-low-power-mode-is-enabled

Setting it is also possible but requires privileges, see https://github.com/jordibruin/cooldown which uses `sudo pmset -a lowpowermode` internally.