exelban/stats #548: 3 Feature requests

johann-lau, 2021-07-19:
Thank you for making this app free and open source. I'd like to submit several requests.
### Bug: fan speed shows 0RPM
The only way to fix this is by switching to a manual speed and switching it back to automatic, though it resets when I restart
### Reorder widgets
It would be great if I can order the widgets. I found a way to do this (disable all widgets, then re-enable them in the correct order), but the sequence is re-ordered when I restart the machine.
### Use more icons from Apple
The app will look a bit more integrated if you use more icons from [Apple](https://developer.apple.com/sf-symbols/).

Thank you for reading to the end!

exelban, 2021-07-19:
Hi. Thanks for the feedback.

Reorder widgets - you can reorder widgets using the default macos mechanism.
SF Symbols - I like this package, but Stats has 10.13 as a minimal macOS version. And SF Symbols 2 requires at least macOS 11.

Fan speed bag. Could you please tell me a little bit more about this? When the value goes to 0? When do you switch to another mode or randomly?

wfrank21, 2021-07-19:
Correction: SF Symbols 2.1 requires 10.14.4

johann-lau, 2021-07-19:
### Monterey
I am using the public beta 3, and it is probably the culprit, so the fan speed bug can be marked as invalid.
### Fan speed bug
It goes to zero when I restart the Mac. When I choose the manual speed option, the speed is shown correctly. However, once I switch to automatic, it will display 0RPM. (Once, it went to ~300 for a second before going down to zero)
[image]
 
### Reorder widgets
Oh, I forgot about the default method :D
### More icons
I know it'd be a pain, but you can actually copy the images like this:
[image]
Then you will get something like this:
[image]
In addition, you can duplicate as a custom icon to tweak the colours. Example (the colours totally make sense):
[image]

exelban, 2021-07-19:
What is the minimal fan speed for your fan? And does this bug appear only on the popup view or also in the menu bar widget?
And you say that it shows 0 RPM, but the fan still works. And after a few seconds, it's turning off. Do I understand correctly?

Ok, but the app does not have any of provided icons. Do you propose to replace some specific icon?
I'm using icons from google material design. So they all have the same design style.

exelban, 2021-07-19:
@wfrank21 are you sure that `SF Symbols` is present on macOS 10.14.4?
I know that an installation package is available. But as I know only from macOS 11 system contains it.

exelban, 2021-07-19:
Also, when you click `Copy Image` you will receive only one image (`35 × 47` for example). Which is not acceptable for icons. It needs at least 3 scales.

wfrank21, 2021-07-20:
@exelban Sorry I was a little bit to fast here, you can install it starting from 10.14.4 but it's only included from macOS 11.

closed 2021-07-25 by exelban