exelban/stats #550: IPv6 Adress shows IPv4

wfrank21, 2021-07-19:
In the current version of stats, the network module popout shows my IPv4 Adress as IPv6.
[image]

**Details:**
 - Device: MacBook Pro 16" 2019
 - MacOS 10.15.7
 - Application version 2.6.1

exelban, 2021-07-19:
Hi. Thanks for the feedback.
There were no changes related to this for a long time.
Could you please check what shows [https://api64.ipify.org](https://api64.ipify.org) (this service is using for IPv6)

wfrank21, 2021-07-19:
It shows the IPv4 Adress.

exelban, 2021-07-19:
Does [https://api.ipify.org](https://api.ipify.org) and [https://api64.ipify.org](https://api64.ipify.org) showing the same ip?

wfrank21, 2021-07-19:
The last group is different.

wfrank21, 2021-07-19:
The correct one is shown on api.ipify.org

exelban, 2021-07-19:
I suppose that it's related to your provider. App has the check if v4 and v6 are the same. To prevent showing v4 in the v6.

wfrank21, 2021-07-19:
Okay you're maybe right. It guess it can be closed.

exelban, 2021-07-19:
But I can prevent showing v4 in the v6 with regex)

closed 2021-07-25 by exelban