exelban/stats #551: Battery % wrong

dg0ojo, 2021-07-19:
[image]

Stats show 1% and Apple 100% :-)

exelban, 2021-07-19:
Hi. Thanks for reporting. I will check this.
It could be fixed at all. Because there are devices that are available only throw macOS cache (`/Library/Preferences/com.apple.Bluetooth`). So the level will be synchronized only when macOS will update them in this file.

Could you please try to restart the app, and check if the battery value will be more correct?

dg0ojo, 2021-07-19:
Hello,

the value was previously 9%. Then I charged the keyboard to 100%. 9% displayed the app correctly.

Mit freundlichen Grüßen
Alexander Laue

mail: ***@***.***
web: https://www.qrz.com/db/DG0OJO

exelban, 2021-07-19:
Emm, sorry but I think I do not understand.
So the app shows the correct battery level?

dg0ojo, 2021-07-19:
The app shows 1% instead of 100%

Mit freundlichen Grüßen
Alexander Laue

exelban, 2021-07-19:
Could you help to understand why it's heppening?

jack24254029, 2021-07-20:
I have a same issue.
The bluetooth device is Magic Trackpad 2.

This is system UI
[image]

This is stats UI
[image]

jack24254029, 2021-07-28:
v2.6.2 still can not show correct battery level of Magic Trackpad 2.
I try to restart the Stats, but not work.

I checked the file (`/Library/Preferences/com.apple.Bluetooth`), I found my Magic Trackpad 2 in Root -> CoreBluetoothCache, and the `BatteryPercent` was `0.65`, but it can not match the battery level on Stats.

[image]

[image]
[image]

jack24254029, 2021-07-28:
I found this.
https://stackoverflow.com/a/59730707/5134328

Maybe can help you.

exelban, 2021-07-28:
Hi. It's strange. Because now my AirPods battery level works fine. And it using the same way. Ok, I will recheck it.

exelban, 2021-07-28:
Could I ask you to check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6894091/Stats.dmg.zip)

jack24254029, 2021-07-29:
No, still show error battery level, but when I reconnect my Magic Trackpad 2, the battery level is updated.

The value diff 1 % between system and Stats, maybe the problem that is carry of the decimal point.

[image]
[image]

jack24254029, 2021-07-29:
I make a situation that charing my Magic Trackpad 2, when I stop charing it, the system UI updated the battery level, but Stats not updated it.
[image]
[image]

I try below solutions:
1. Restart Stats -> NOT UPDATED
2. Reconnect Magic Trackpad2 -> UPDATED

jack24254029, 2021-07-29:
I charing my Magic Trackpad 2 to 100%, Stats show 1%.

[image]
[image]

This is my device cache
[image]

exelban, 2021-07-29:
Hi. Thanks for such a nice debug. According to the second message, the problem is not in the Stats. But in the process of how cache in macOS is updating.
I suppose the value in the cache is updated only on some trigger. In the case of the trackpad is reconnect. 

I notice the same bug with my AirPods. It updates the battery level on reconnect. I will try to fetch the battery level from `ioreg`. Maybe it contains more "live" data.

exelban, 2021-07-29:
And the `1%` when cache contains `1.00` could be related to the float/int conversation. Because the same value in some cases is Int and in some Float. I will fix it.

exelban, 2021-07-30:
Hi. I take a look at the `ioreg`. It's not possible to fetch a BLE device better level from it. Maybe, it's acceptable for a few devices ( don't have any of them). But it's not a universal solution. I don't find a better way than `com.apple.Bluetooth`.

closed 2021-08-01 by exelban

jack24254029, 2021-08-02:
Hi @exelban 
I search the method to get the battery level of Magic Trackpad2.
I wrote the sample code on Playground.
Hope can help you.

```Swift
import Foundation
import IOKit

var serialPortIterator = io_iterator_t()
var object: io_object_t
let masterPort: mach_port_t = kIOMasterPortDefault
let matchingDict: CFDictionary = IOServiceMatching("AppleDeviceManagementHIDEventService")
let kernResult = IOServiceGetMatchingServices(masterPort, matchingDict, &serialPortIterator)
if KERN_SUCCESS == kernResult {
    object = IOIteratorNext(serialPortIterator)
    while object != 0 {
        if let product: Unmanaged<AnyObject> = IORegistryEntryCreateCFProperty(object, "Product" as CFString, kCFAllocatorDefault, 0) {
            let productValue: String = product.takeRetainedValue() as! String
            print("Product: \(productValue)")
        }
        if let batteryPercent: Unmanaged<AnyObject> = IORegistryEntryCreateCFProperty(object, "BatteryPercent" as CFString, kCFAllocatorDefault, 0) {
            let batteryPercentValue: Int = batteryPercent.takeRetainedValue() as! Int
            print("Battery Level: \(batteryPercentValue)")
        }
        object = IOIteratorNext(serialPortIterator)
    }
    IOObjectRelease(object)
}

IOObjectRelease(serialPortIterator)
```
Result
[image]

I don't have a bluetooth earphone, so I can't test it.

exelban, 2021-08-02:
@jack24254029 Thanks. As I said previously, it will work only with some devices. This way works only because Magic Trackpad is adding to the system as an input device (and only Apple devices, because my BLE keyboard does not visible in the `AppleDeviceManagementHIDEventService`). So it could be fetched via `ioreg`. Another device will not work with this way of obtaining the battery level. So it could not be used to get BLE battery levels.

I propose you continue this discussion in #557.
Because maybe it will be the only way to obtain Magic mouse/keyboard levels. And I will add this method for specific devices.

jack24254029, 2021-08-02:
> @jack24254029 Thanks. As I said previously, it will work only with some devices. This way works only because Magic Trackpad is adding to the system as an input device (and only Apple devices, [...]

OK, thank you.

leobrafl, 2022-06-20:
Not sure if this is still here, but I'm not being able to see my Bluetooth mouse (Apple Magic Mouse) on the dashboard there
However, when I check iStats, which I was previously using, it shows the battery level of my Bluetooth mouse normally. I dropped some images to show that, even though it's recognizing the mouse, it's not showing the battery level

[image]
[image]