exelban/stats #552: CPU Top Processes refreshes too fast

godly-devotion, 2021-07-20:
**Describe the bug**
The Top Processes list does not seem to respect the update interval option. Regardless of setting the value to 1 second or 10 seconds, the list always refreshes every second. This makes it very hard to scan the top processes list for me because it refreshes too fast.

**Details:**
 - Device: MacBook Pro 13" (Mid 2015)
 - MacOS 11.4
 - Application Version 2.6.1

exelban, 2021-07-20:
Hi.
`Update interval` option is acceptable only for the main reader (CPU/RAM/etc usage). It does not affect the `top processes`

godly-devotion, 2021-07-20:
I see. I expected the `top processes` to follow the update interval since everything else in the widget view, the pie chart, history graph, and System/User/Idle info was also following the update interval. It feels natural that the `top processes` should also follow the users update interval choice. Is there perhaps a technical limitation to make it consistent?

exelban, 2021-07-20:
There is, and at the same time there are no technical limitations)
But I device 2 types of `readers`: main and additional. Main - it's a reader of the main resource (CPU/RAM/etc usage). Additional - it's a `top`, CPU temperature and frequency readers.

The difference between them: one (main) is running all time, and the other (additional) only when it's needed (opened popup view).

Why intervals cannot be the same? Because for example, I want to have RAM usage updates every 3s (because it's not changing fast). But `top processes` - every second. Because it's important to me to see the live update by processes. 

There is an option of how it could be resolved: add an `additional reader update interval`.
But I think it's a super complicated option, and will be totally not obvious for users.

godly-devotion, 2021-07-20:
Agreed, having another update interval seems overcomplicated. However it does bring back the original issue which the `top processes` are updating a bit too fast for me to read it. This could be a bit more complicated, but perhaps the frequency reader code be refactored to allow the consistent behavior? I do think this would make the update interval more consistent. However if it's too difficult, it is what it is.

exelban, 2021-07-20:
As I say, I'm not sure that the same update interval is a good solution. Because there are more cases when it must be different. Also, it could be implemented in the CPU and RAM modules only. It cannot be realized in the network top processes at all (I hope this doesn't need the explanation why). And battery module which is using `top` underhood cannot have this option based on the `top` recommendation. Also, experience says that in the battery module it's not critical at all.

Hmm, maybe it could be realized as a `Top processes update interval`, but in the app settings. Not in each reader. But again, this option will be available only for CPU and RAM top. That could mislead users. 

I understand the problem (having the same from time to time). But I personally using these `top processes` for a quick look at resource usage. I open the Activity Monitor when I need to debug something or track some app.

godly-devotion, 2021-07-20:
> Hmm, maybe it could be realized as a `Top processes update interval`, but in the app settings. Not in each reader. But again, this option will be available only for CPU and RAM top. That could mislead users.

That sound like a good idea. Perhaps maybe `Top processes update interval for CPU/RAM` to avoid confusion.

exelban, 2021-07-20:
Nope, `Top processes update interval for CPU/RAM` looks very strange in the main settings. And also it's a too long title which will not fit(

Maybe, I'm not sure, but maybe.
I could add a pause/resume button in the popup view. Which will stop the updates of `top processes`. It could be useful if you want to debug something. But still, I recommend using this list only as a top. And using Activity Monitor for debugging/tracking.

godly-devotion, 2021-07-20:
Sorry if I'm coming off as too insistent haha... What about like this?
[image]

exelban, 2021-07-20:
It's the first option that we are talking about. In theory, it could be added. I need to check if it's ok to change the update interval of `top processes` readers at all (in Stats, not `ps`).

closed 2021-08-01 by exelban