exelban/stats #553: [Feature Request] RAM/CPU: Ability to show most resource intensive process in menu bar

godly-devotion, 2021-07-20:
The ability to show RAM/CPU percentage is great way to see if there is an offending process that shouldn't be taking up so much resources. It would be really helpful to have an option to see which process is consuming the highest CPU (or RAM if thats the menu bar module) right on the menu bar at a glance. Ideally it would respect the update interval option so that the user has control as to how often the process name gets updated in the menu bar. Thanks!

exelban, 2021-07-20:
Hi. Thanks for the proposal.
But I'm not sure that it's a good feature. 

First of all because continuously reading the `top` processes will load the CPU a lot. 
Second, I can't even imagine how it could be implemented from UI/UX side. Showing the PID in the menu bar? Showing the process name in the menu bar? Additionally, there must be some minimal level of use. Because you don't want to see the most using process which is using 0.1% of CPU. What if the system does not have a process that is higher than the limit? Showing empty space in the menu bar? Or design the jumping menu bar?

I don't even see the option of how it could be implemented.

godly-devotion, 2021-07-20:
Thanks for the quick reply! You have some good points there.

From the UI perspective, rather than showing the top process always, it may be better to have a percentage limit as you've mentioned that can also be user customizable. So that you only see those over 50% cpu utilization for example. I usually put the stat menu items at the end so I'm not terribly concerned about the text label length changing but if having a blank label is weird (because no processes met the threshold), it could be an option that can show with the CPU percentage option so that it is anchored to something (since CPU percentage would always show). As for the continuous `top` process utilization I don't know if that is a real issue for most modern macs. For example, running the `htop` process ends up being only 0.7% cpu utilization on my i5 2015 MacBook Pro. However there could be a label on the option that says that it could take some cpu usage to display the top process name to inform the user.

exelban, 2021-07-20:
Jumping menu bar - I mean that this widget will have an adjustable size. And when there is no text, the width will be 0.
Sorry, but it cannot be an empty space in the menu bar at all. 

CPU utilization for me it's an important moment. And I will not add a feature that will load the CPU more than the whole app (continuously). And it cannot be a feature that could be disabled if the widget is not enabled (technical limitation).

For me, it looks more like a feature for a separate app.

godly-devotion, 2021-07-20:
> Jumping menu bar - I mean that this widget will have an adjustable size. And when there is no text, the width will be 0.
> Sorry, but it cannot be an empty space in the menu bar at all.

We could make it so that it's an info that can be shown with the CPU percentage option so that it would never be empty.
Example:  [ 40% ] or if CPU is above threshold [ Safari - 60% ]

> CPU utilization for me it's an important moment. And I will not add a feature that will load the CPU more than the whole app (continuously). And it cannot be a feature that could be disabled if the widget is not enabled (technical limitation).

Reading the other ticket with the architecture of the readers, yeah I can see how it would be difficult to implement that way.

If its not possible, thats fine. The app is pretty amazing as is already 🙂

closed 2021-07-20 by godly-devotion

exelban, 2021-07-20:
`[ 40% ]` and `[ Safari - 60% ]` have different width. And if in one moment the width is 10px and in another 20px. It means that all components in the menu bar will jump because this width was changed. 
Additionally, it (I mean this text - `Safari) cannot be just added to the widget. It requires a new widget. And I totally will not add an additional widget for feature like this. Because it's a super rare feature (see the most using process) which will require the support of the new widget.