exelban/stats #554: Mac stopped charging today (Potential SMC problems)

cristiean, 2021-07-20:
My mac stopped charging today. Stats is the only software I use that plays around with the SMC.

Last night I spun up the fans with the new Turbo mode.

In no way I am accusing Stats of leading to this problem. But I wanted to share this in case anyone else experiences it and it's potentially caused by Stats.

Maintainer, you're doing an awesome job at building this project. This is in no way an attack but a way of potentially helping other people in case Stats is the issue. If it is, I'm sure it will be fixed.

**Details:**
 - Device: Macbook Pro 13 2018
 - MacOS big sur

cristiean, 2021-07-20:
After restarting it a few times and resetting the SMC, it now seems like it's taking charge again. What a scare!

closed 2021-07-20 by cristiean

exelban, 2021-07-20:
Hi.
First of all. It's good not to use tools if you don't know what they doing. Manipulation with SMC it's not a good solution at all. If Apple will ok with manipulating with SMC, they will make an API for this. But there is no option.
You decided that you know better than macOS which fan speed is better. And manually change the speed. 

The second. I add limits to the fan speed, which are recommended by apple (minimal and maximum). So it's safe to set max speed because macOS using the same maximum value for fan speed when needed. So you cannot go higher (or lower) than designed by Apple.

But still, I do not recommend using manual fan control. But if you decided to, please understand that something could go wrong. There is this feature in the app because there were many requests for this. And I'm pretty sure that it could not break mac (ok there are some cases, for example, if your fan for a long time does not spin up for max speed, and suddenly you will manually spin it).

Additionally, there are many checks in SMC. For example, if you manually change the fan speed to 0, the fan will not stop. SMC will set the lowest value.