exelban/stats #556: Dropdown panels not showing or switching screens when using full screen apps

GTavanez, 2021-07-22:
**Describe the bug**
Attempting to open one of the dropdown panels, when using a full screen app, will cause the screen to slide to a workspace with nom maximised windows, or switch focus on an available workspace with non maximised windows in case multiple monitors are in use.
Example video Included
https://user-images.githubusercontent.com/40792564/126638359-4e2bdcd5-fcf0-4e39-813c-539dfe39f22d.mp4

**Details:**
 - Device: MacBook Pro (Retina, 15-inch, Mid 2015)
 - MacOS 10.15.7 (19H1217)
 - Application version: 2.6.1

exelban, 2021-07-25:
Hi. Thanks for repoting. 
I will take a look at this. I remember that it was reported already something similar to this issue. And it could not be fixed. But I will take another try.

GTavanez, 2021-07-26:
Was not aware of that, so thank you for giving it a second look! Let me know if you need some testing, settings, anything

exelban, 2021-08-03:
@GTavanez Hi. Could you please tell me what connection type (Type-C/HDMI/DP) did you use for the monitor?

GTavanez, 2021-08-03:
@exelban hi! I'm using Hdmi on a Dell U2715H, max res (2560 X 1440).
But that's not really relevant unless you have a full screen app in one monitor, and a non full screen in the other, that's when it changes screen, otherwise, with a single screen, it will just move workspace

exelban, 2021-08-03:
Unfortunately, I cannot reproduce this bug. I tried HDMI and type-c connections. I also try different ways of monitor settings (macbook main / monitor main). And everything work just fine.

GTavanez, 2021-08-03:
Having an external monitor isn't really relevant, more of a side twist to the actual "issue".
As you can see in my video, it happens with a single screen, you just need to maximise an app, and when clicking on details to open the dropdowns, it will switch to a non maximized workspace

exelban, 2021-08-03:
I suppose it's related to some settings of macOS. Because it's only the second issue with this behavior. I have a mac's with 10.15.7 and 11.3. And any does not have this behavior. 

Unfortunately, I cannot fix it without the option to reproduce this behavior.

GTavanez, 2021-08-03:
I fully understand that, if when you're in a full screen app, the dropdown shows correctly, then I'll try to do some further troubleshooting on my end, and let you know what I find, perhaps it can help someone in the future

exelban, 2021-12-05:
Hi. For all this time I have never been seen this bug. I suppose it's related to your monitor or maybe something in your mac. I don't see a way to reproduce this bug and fix it.

closed 2021-12-05 by exelban