exelban/stats #557: Magic Keyboard 2 & Magic Mouse Battery not showing up

wfrank21, 2021-07-23:
I don't know if Magic Keyboard 2 and Magic Mouse are supported by the BT Module, but if so, they are not showing up on my Mac. AirPods Pro do. It could be, that my config (iMac 2011, Big Sur, OpenCore Bootloader), is the issue.

**Details:**
 - Device: iMac 27" mid 2011
 - MacOS 11.5
 - Application version 2.6.1

exelban, 2021-07-23:
Hi. Thanks for reporting.
I will need to find a keyboard and a mouse to debug it. It will take some time.

wfrank21, 2021-07-23:
If you need any help, let me know.

exelban, 2021-07-23:
Yes, you can help. There is a file `/Library/Preferences/com.apple.Bluetooth`. Could you open it and find the `DeviceCache` field. Inside this field will be the list of Bluetooth devices. Could you find your keyboard and mouse and post them?

wfrank21, 2021-07-23:
I'm currently at work, I will post them when I'm at home again (today in the afternoon)

wfrank21, 2021-07-23:
Magic Keyboard:
`<key>DeviceCache</key>
	<dict>
		<key>10-94-bb-bf-50-5a</key>
		<dict>
			<key>BatteryPercent</key>
			<real>0.80000001192092896</real>
			<key>ClassOfDevice</key>
			<integer>1344</integer>
			<key>ClockOffset</key>
			<integer>0</integer>
			<key>ColorID</key>
			<integer>5</integer>
			<key>EncryptionKeySize</key>
			<integer>170</integer>
			<key>ExtendedFeaturesPage1</key>
			<data>
			AAAAAAAAAAE=
			</data>
			<key>HeySiriEnabled</key>
			<integer>0</integer>
			<key>LMPSubversion</key>
			<integer>9228</integer>
			<key>LMPVersion</key>
			<integer>5</integer>
			<key>LastInquiryUpdate</key>
			<date>2021-07-05T18:10:58Z</date>
			<key>LastNameUpdate</key>
			<date>2021-05-30T07:20:05Z</date>
			<key>LastServicesUpdate</key>
			<date>2021-07-02T11:25:46Z</date>
			<key>Manufacturer</key>
			<integer>15</integer>
			<key>Name</key>
			<string>wfrankMagicKeyboardBlack</string>
			<key>PageScanMode</key>
			<integer>0</integer>
			<key>PageScanPeriod</key>
			<integer>0</integer>
			<key>PageScanRepetitionMode</key>
			<integer>1</integer>
			<key>ProductID</key>
			<integer>620</integer>
			<key>SerialNumber</key>
			<string>F0T8496004THTGMAM</string>
			<key>ServiceMask</key>
			<integer>32</integer>
			<key>Services</key>
			<data>`

Magic Mouse

`		<key>a8-91-3d-39-1f-d6</key>
		<dict>
			<key>BatteryPercent</key>
			<real>0.94999998807907104</real>
			<key>ClassOfDevice</key>
			<integer>9600</integer>
			<key>EncryptionKeySize</key>
			<integer>170</integer>
			<key>ExtendedFeaturesPage1</key>
			<data>
			AAAAAAAAAAE=
			</data>
			<key>HeySiriEnabled</key>
			<integer>0</integer>
			<key>LMPSubversion</key>
			<integer>9228</integer>
			<key>LMPVersion</key>
			<integer>5</integer>
			<key>LastNameUpdate</key>
			<date>2021-05-30T07:20:05Z</date>
			<key>LastServicesUpdate</key>
			<date>2021-07-02T11:25:38Z</date>
			<key>Manufacturer</key>
			<integer>15</integer>
			<key>Name</key>
			<string>wfrankMagicMouseBlack</string>
			<key>ProductID</key>
			<integer>617</integer>
			<key>SerialNumber</key>
			<string>CC2046206V8J51XA8</string>
			<key>ServiceMask</key>
			<integer>32</integer>
			<key>Services</key>
			<data>`

exelban, 2021-07-23:
Thanks for sharing. Looks like it must be visible. I will check the apple mouse the next week and will give to know.

exelban, 2021-08-02:
@jack24254029 Could you maybe share the full object of the device from `AppleDeviceManagementHIDEventService`?

jack24254029, 2021-08-03:
> @jack24254029 Could you maybe share the full object of the device from `AppleDeviceManagementHIDEventService`?

Sure.
[image]

closed 2021-08-15 by exelban